prusa3d/PrusaSlicer #2991: Probleme memoire

Kiffou, 2019-09-25:
[image]
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2019-09-25:
please provide the 3mf file and the sequence to reproduce your issue.

forestqc69, 2019-09-26:
I think I have the same problem here, Memory usage goes to 6700MB then crashes on bad alloc, app has to be force closed or it stays in idling status.

My system : Win 10 (18362) 64bit, 12592MB RAM
Slicer: 2.1.0+win64-201909160915

Problem seems to be that the 80% infill is too much, lower settings works. I see a spike in memory usage, a drop, then a steady climbing again until it crashes when system is running out of available memory..

See attached file for .3mf

Thank you!

[Hurricane.zip](https://github.com/prusa3d/PrusaSlicer/files/3655679/Hurricane.zip)

Kachidoki2807, 2019-10-01:
I gave a try for fun. No need for tricky settings or model. Took a simple xyz calibration cube, standard 0.20mm quality settings, then I scaled it up to full volume (F shortcut in scale gizmo). The slicing leads to a RAM allocation >19GB while processing. With the other program running, I am near the limits (but not reached yet, 95%) of my 32GB computer.

bubnikv, 2019-10-02:
80% gyroid infill over the whole build space?
The biggest memory hog is the 3D print path preview.

Kachidoki2807, 2019-10-02:
My test with 19GB of RAM used was with 15% gyroid infill (stock settings).
Even a 15% honeycomb infill take "only" 4GB.
But indeed, switching to a 80% cubic infill take only 1GB of RAM in comparison.

Now that the full scale feature (#1509) is implemented, the users who want to use it will easily get into this trap. Maybe a warning should appear, or at least, switch the infill type to the "old' cubic one.

bubnikv, 2020-03-15:
PrusaSlicer 2.2.0-rc3 should be more conservative when slicing gyroid infill. Does that resolve your issue?

Kachidoki2807, 2020-03-16:
Hi bubnikv, I have done the exact same test as before (xyz cube at full scale, stock settings) on the version 2.2.0-rc4. It now consume a bit less of RAM, up to ~14GB.

I notice that this is the last step of the slicing process which take sooo long time (running only on one core) and massively increase the RAM usage. I notice also a relative high (~12MB/s) SSD usage at this moment.

To summarize:
- Before this last step, the slicing take few seconds and the memory usage rise up to ~8-9GB.
- During this step, the memory consumption rise up to ~14GB.
- After this last step which take several minutes, the RAM usage drops from to ~9GB.

bubnikv, 2020-03-16:
the most memory consuming is the 3D print path preview - for every path
segment there are 8 triangles, minimum 4 vertices with normals(4x3x4x2
bytes) generated.

Kachidoki2807, 2020-03-16:
Sure, I understand the huge RAM usage, sometime it is not possible to optimize more, and of course, the gyroid infill with a lot of small arcs, handled as even more smaller straight segments is almost the worst case.

But what I did notice, it's that the first steps of the slicing process (before the G-Code generation) already produce an usable output, very quickly (before the feature color changes), and with less RAM usage. When your are doing a lot of small parameter changes to optimize the slicing, this first output is enough to check the impact of theses small changes. Maybe a checkbox option somewhere could insert a step into the interface workflow like "Slice now -> Generate G-Code -> Export G-Code". It will save a lot of time during the parameters tuning, and before the real complete slicing. I also think that it is outside the scope of this issue.

Of course, this is a specific case, most of the time it will not be annoying.

bubnikv, 2020-03-16:
We are investigating excessive memory consumption by the print time
estimator. We will likely decrease memory consumption in PrusaSlicer 2.3 to
some extent

rtyr, 2023-09-12:
No response. I am closing the issue.