prusa3d/PrusaSlicer #2993: Support enforcer not fully added

hetfmt, 2019-09-25:
### Version
PrusaSlicer Version 2.1.0+win64

### Operating system type + version
Windows 10 Home
Version 10.0.18362 Build 18362

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3
Firmware 3.8.0

### Behavior
Regardless of whether I use "Supports:"
* Everywhere
* Support on build plate only
* For support enforcers only - with custom support added
 No support is added for the large overhanging perimeter (divider of upper and lower half). From the legend, it looks like it is being interpreted as "Bridge infill", where I would certainly consider it to be overhanging. Only support for the holes in the side is added.
If I turn the drawing up-side down, the problem persists.

#### Project File (.3MF) where problem occurs
[kopholder.zip](https://github.com/prusa3d/PrusaSlicer/files/3654222/kopholder.zip)
[image]
[image]
[image]

[kopholder.zip](https://github.com/prusa3d/PrusaSlicer/files/3654245/kopholder.zip)

_Is this a new feature request?_ No.

BerndJM, 2019-09-25:
If the slicer recognize (for whatever reason) such overhangs as bridges, just uncheck "Don't support bridges" in Print Settings -> Support material ...

neophyl, 2019-09-25:
That's because it is bridging.  With properly tuned printer settings it is actually possible to print that without support with bridging.  At least in pla.  If you really want to force supports under there you must disable bridging detection.  Print Settings > Layers & perimeters and untick detect bridging perimeters.

I'd try it first without any supports, it might just surprise you.

hetfmt, 2019-09-26:
Excellent! Thank you, both of you. I will try your suggestions.
I am printing with PETG, and though the print is successful without support, the resulting finish of the top of the "bridge" is not nice. 
Thinking about it: maybe if I set infill to 100% for the "bridge" (entire dividing plate) it might produce a decent result. I think the problem might be related to adding a low level of infill on a "floating layer"?

closed 2019-09-30 by hetfmt