prusa3d/PrusaSlicer #2994: "M104 S182 ; set temperature" Added in G-code 2nd layer

AbeFM, 2019-09-26:
### Version
2.1.0+

_Use `About->About PrusaSlicer` for release versions_

### Operating system type + version
win10

### 3D printer brand / version + firmware version (if known)
mk3s - N/A

### Behavior
Slice code, using Mk3s or Mk3S+MMU2 - at 0.15 layer height with 0.2 mm first layer (probably numbers don't matter but didn't test yet)
Right after the ;0.35 height marker, I see an M104 command.
```
G1 Z0.350 F10800.000
;AFTER_LAYER_CHANGE
;0.35
M104 S182 ; set temperature
G1 X154.726 Y93.842
G1 E1.50000 F2100.00000
G1 F2400

```
I'm not quite sure what it is, and it's different depending on what I'm printing. It's likely first layer temp for one particular material, but I just wanted to get this out there so I'm posting what I know.

#### Project File (.3MF) where problem occurs
I've seen this across different projects - if this isn't trivial to reproduce ask and I'll provide more info.

lukasmatena, 2019-09-26:
> ask and I'll provide more info.

I'll ask then: Would you be so kind to provide a 3MF, please? There is a very good reason we ask for it in the issue template that people tend to just delete and ignore. Please, provide a 3MF. Thanks.

AbeFM, 2019-09-26:
Thanks for being understanding, it was the middle of the night and I had work to get done.

[Line392.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/3659734/Line392.3mf.txt)

Mk3S and MMU2S Single profiles didn't have this issues.

lukasmatena, 2019-09-26:
Thanks. After slicing your 3MF, exporting gcode and looking at line 392, I found this. The temperature is set to 205 degrees, exactly as demanded by Filament Settings.
[image]

I don't know where the 182 degrees you mention in your first post came from, but since second layer is at 0.35mm in your first post and at 0.4mm in your 3MF, clearly those are two different scenarios.

Shortly put, the 3MF you provided works exactly as it should. If there is a case where it is otherwise, please, provide a 3MF with that scenario so we can reproduce it.

AbeFM, 2019-09-26:
Thanks... AGAIN. I need to dig into this until I've figured out what's going on. I checked 3 old g-codes and found issues, but haven't tracked it down yet.

closed 2019-09-26 by AbeFM