prusa3d/PrusaSlicer #2995: Adding setting/modifier to MM part makes it impossible to set extruder

xiphmont, 2019-09-26:
### Version
2.1.0

### Operating system type + version

Linux (Fedora 31 w/Wayland)

### Behavior
Once a setting has been added to a multimaterial part, it is no longer possible to change the extruder setting on any part.  Clicking on the extruder number initiates a drag of the part instead  of opening the extruder number dropdown

_Is this a new feature request?_
No

Illustrative 3mf file attached-- extruders can be set/changed on the Plater screen  in PrusaSlicer 2.0 but not 2.1.

[flask.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/3657313/flask.3mf.gz)

Jebtrix, 2019-09-26:
I can't reproduce this on my Arch box. Repo, appimage, or tarball? All my boxes have nvidia so I'm not up to speed on possible Wayland induced issues. If you temporarily turn off wayland is it any different in X11? ie `WaylandEnable=false in /etc/gdm/custom.conf`

Jebtrix, 2019-09-26:
Fedora 31 using wayland inside a VM worked fine for me using v2.1.0 appimage.

xiphmont, 2019-09-26:
Thank you for the *very* quick triage.

I forgot something very important and obvious-- this was a local build from source.  The 2.1.0 AppImage works flawlessly, so there's some problem with a local dependency, perhaps the version of wx I'm using.  I'll continue tracking it down here.

Is there an easy way to see the dependencies the AppImage was built with?

Jebtrix, 2019-09-26:
Their appimage is built against wxwidgets v3.1. If you didn't use wxwidgets dev branch to get 3.1.xx then that would be the silver bullet. Trying to see dependencies from a static linked binary is a manual and most likely futile process. Add the symbol and debug stripping your not left with much. Readelf doesn't even recognize the binary as a proper elf format once you extract it from the appimage.

bubnikv, 2019-09-27:
The trouble with wxWidgets is complexity of testing due to
muti-platformness and dependency on platform UI tollkits, a relatively slow
pace of development and testing, and that the linux world adheres to the
versions declared as stable, which as of now is the 2.0 series. We at the
PrusaSlicer team are looking forward for the wxWidgets team to declare the
3.2, but according to
https://trac.wxwidgets.org/wiki/Roadmap
"New development happens on 3.1 branch. The latest version is 3.1.2,
released on December 10, 2018 and we plan to release 3.1.3 in October, with
3.1.4 following in the beginning of 2020 and 3.2.0 hopefully soon
afterwards."

Therefore we have to somehow support the wxWidgets 3.0 on Linux distros,
but we do not have resources for that. We squash some bugs in PrusaSlicer
due to wxWidgets 3.0 incompatibility with wxWidgets 3.1 occasionally, but
we cannot declare wxWidgets 3.0 as fully supported by us. We will certainly
be very thankful for pull requests for wxWidgets 3.0 compatibility until
wxWidgets 3.2 is declared stable.

Also due to various reasons, we maintain our own branch of wxWidgets 3.1.1
with some features from newer versions of wxWidgets back ported to our
branch of wxWidgets 3.1.1. We build PrusaSlicer statically agains this
wxWidgets branch on all platforms, you can do the same too.

Jebtrix, 2019-09-27:
@Salamandar take note here. If you are the same Salamandar that maintains prusa-slicer-git in AUR why don't you adjust the makepkg dependencies for static linking for wxwidgets 3.1 by default. It's not against package guidelines and I'm pretty sure arch users would appreciate official appimage/tarball build equivalency. As a side note,

Salamandar, 2019-09-27:
@Jebtrix I'll look into it. But static linking is not a great idea…
Anyways, tl;dr what's the issue here ?

xiphmont, 2019-09-28:
Still working on verifying that wx3.1 will in fact eliminate the proble in a local build.

wx 3.1 won't build here on F31-- the cmake setup keeps missing include subdirs, and I'm unfamiliar with cmake, so that's slowing me down.

xiphmont, 2019-09-28:
Confirmed-- built against a local checkout of the Prusa wx 3.1 fork with no other changes, the UI behaves as expected (both item drag and extruder number dropdown work properly).  So the problem was definitely wx 3.0.

I'm satisfied with this resolution.  Close at your leisure.

bubnikv, 2019-09-30:
I am leaving it open for reference. I tagged this issue with a "wxWidgets 3.0" tag.

lukasmatena, 2023-11-21:
Closing as outdated.

closed 2023-11-21 by lukasmatena