prusa3d/PrusaSlicer #2996: Layer inverted when slicing some objects, making them unprintable

adamroach, 2019-09-26:
### Version
2.1.0+

### Operating system type + version
macOS 10.13.6

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* On some models, the slicer will "reverse" layers (putting filament where voids should be, leaving voids where filament should be.
* _Steps needed to reproduce the problem_
  * Open the attached STL file in Prusa Slicer: [Thor+Door+Stop.zip](https://github.com/prusa3d/PrusaSlicer/files/3660268/Thor%2BDoor%2BStop.zip)
  * If you want to generate your own or view the source, this comes from https://www.thingiverse.com/thing:3022244
  * Slice the object.
  * Observe that several layers are reversed, rendering the object unprintable
* _Expected Results_
  * The object should be printable.
* _Actual Results_
  * The object is not printable.

_Images_
The object is hollow all the way through to the plate:
[image]

When sliced, there is an inverted layer added:
[image]

[image]

[image]

[image]

Note that this model works just fine in the original Slic3r code:
[image]

_Is this a new feature request?_
No.

adamroach, 2019-09-26:
Here is another example: [ukelele-top.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3660307/ukelele-top.3mf.zip)

You'll see that the part should be hollow all the way down to the plate; but, when sliced, Prusa Slicer inserts an incorrect solid layer that makes it unprintable:

[image]
[image]

[image]
[image]

Jebtrix, 2019-09-27:
Both meshes have multiple mesh problems from non manifold to zero length edges and zero area faces.  Most people will use netfabb to attempt repair.

Jebtrix, 2019-09-27:
Cleaned em up quick in Blender to slice fine though some thin and intersecting faces remain in the one. 

[door_wedge_20190926-56-1wscgw1_cleaned.zip](https://github.com/prusa3d/PrusaSlicer/files/3660568/door_wedge_20190926-56-1wscgw1_cleaned.zip)

[top-maybe_cleaned.zip](https://github.com/prusa3d/PrusaSlicer/files/3660570/top-maybe_cleaned.zip)

The fact remains Cura and Simplify3D managed to slice these acceptably regardless of problems though. Devs mentioned STL importing will be getting some collective focus so we just have to be patient.

Jebtrix, 2019-09-27:
Fixing manifold problems certainly is. I've stated before the ability for slicers to handle garbage meshes impresses me. I'm just stating observations, nothing more.

tkircher, 2019-09-29:
Really, Slic3r should have a pop-up any time an imported model isn't manifold, telling the user that their mesh isn't valid.

bubnikv, 2019-09-30:
We show exclamation marks in the object list left of each item that
PrusaSlicer has issue with. If you are on Windows 10 and you click on the
exclamation mark, the model is sent to Microsoft hosted Netfabb service for
fixing.

tkircher, 2019-09-30:
They don't seem to be visible enough, considering how many people file a bug report only to discover later that the problem is their model. Why does it even allow users to slice a model that's broken?

adamroach, 2019-10-03:
> We show exclamation marks in the object list left of each item that PrusaSlicer has issue with.

Well, except these specific models that I attached to this issue don't have those exclamation points when you load them.

closed 2026-09-29 by prusaslicer-bot[bot]