prusa3d/PrusaSlicer #2997: App crash on shift + mouse clicking empty space in STL list

jboymel, 2019-09-27:
### Version
2.1.0+win64

### Operating system type + version
Win10

### Behavior
Load any STL, select the STL in the list on the right menu.
Hold down shift key and left or right click on empty space in STL list.

[image]

Jebtrix, 2019-09-27:
I can reproduce it on Win7, Win10 and partially on Linux (only shift+rightclick). 

Backtrace:

```
#0  0x0000000000cae56b in wxEventFunctorFunctor<wxEventTypeTag<wxDataViewEvent>, Slic3r::GUI::ObjectList::ObjectList(wxWindow*)::{lambda(wxDataViewEvent&)#1}>::operator()(wxEvtHandler*, wxEvent&)   
    ()
#1  0x00000000011f0455 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (entry=..., handler=<optimized out>, event=...) at ./src/common/event.cpp:1396
#2  0x00000000011f0869 in wxEvtHandler::SearchDynamicEventTable(wxEvent&)
    (this=this@entry=0x2748e00, event=...) at ./src/common/event.cpp:1868
#3  0x00000000011f096e in wxEvtHandler::TryHereOnly(wxEvent&) (this=0x2748e00, event=...)
    at ./src/common/event.cpp:1589
#4  0x00000000011f0a16 in wxEvtHandler::TryBeforeAndHere(wxEvent&) (event=..., this=0x2748e00)
    at ./include/wx/event.h:3873
#5  0x00000000011f0a16 in wxEvtHandler::ProcessEventLocally(wxEvent&) (this=0x2748e00, event=...)
    at ./src/common/event.cpp:1526
#6  0x00000000011f0a94 in wxEvtHandler::ProcessEvent(wxEvent&) (this=0x2748e00, event=...)
    at ./src/common/event.cpp:1499
#7  0x00000000011f1417 in wxEvtHandler::SafelyProcessEvent(wxEvent&)
    (this=<optimized out>, event=...) at ./src/common/event.cpp:1617
#8  0x000000000103085c in wxWindowBase::HandleWindowEvent(wxEvent&) const (this=this@entry=0x2748e00, event=...) at ./src/common/wincmn.cpp:1539
#9  0x0000000000ee63ec in wxdataview_selection_changed_callback(GtkTreeSelection*, wxDataViewCtrl*) (dv=0x2748e00) at ./src/gtk/dataview.cpp:4427
#10 0x00007ffff6a98caa in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#11 0x00007ffff6a86235 in  () at /usr/lib/libgobject-2.0.so.0
#12 0x00007ffff6a8a89a in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#13 0x00007ffff6a8b700 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#14 0x0000000000ee78a8 in gtk_dataview_button_press_callback(GtkWidget*, GdkEventButton*, wxDataViewCtrl*) (gdk_event=<optimized out>, dv=0x2748e00) at ./src/gtk/dataview.cpp:4564
#15 0x00007ffff723b7cc in  () at /usr/lib/libgtk-x11-2.0.so.0
#16 0x00007ffff6a98caa in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#17 0x00007ffff6a86706 in  () at /usr/lib/libgobject-2.0.so.0
#18 0x00007ffff6a89e2c in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#19 0x00007ffff6a8b700 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#20 0x00007ffff7356235 in  () at /usr/lib/libgtk-x11-2.0.so.0
#21 0x00007ffff7239a0e in gtk_propagate_event () at /usr/lib/libgtk-x11-2.0.so.0
#22 0x00007ffff7239e43 in gtk_main_do_event () at /usr/lib/libgtk-x11-2.0.so.0
#23 0x00007ffff6eb2d5e in  () at /usr/lib/libgdk-x11-2.0.so.0
#24 0x00007ffff69adcf4 in g_main_context_dispatch () at /usr/lib/libglib-2.0.so.0
#25 0x00007ffff69afb11 in  () at /usr/lib/libglib-2.0.so.0
#26 0x00007ffff69b0a63 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#27 0x00007ffff7238df3 in gtk_main () at /usr/lib/libgtk-x11-2.0.so.0
#28 0x00000000010761a5 in wxGUIEventLoop::DoRun() (this=0x73c7f40) at ./src/gtk/evtloop.cpp:65
#29 0x000000000111aa22 in wxEventLoopBase::Run() (this=0x73c7f40) at ./src/common/evtloopcmn.cpp:76
#30 0x00000000010ebeb0 in wxAppConsoleBase::MainLoop() (this=0x1e9c930) at ./src/common/appbase.cpp:380
#31 0x00000000011687c4 in wxEntry(int&, wchar_t**) (argc=<optimized out>, argv=<optimized out>) at ./src/common/init.cpp:507
#32 0x00000000008ff5e2 in Slic3r::CLI::run(int, char**) ()
#33 0x00000000008c81bb in main ()

```

DominikPalo, 2019-09-28:
I can reproduce it also on macOS 10.14.6

Backtrace:
```
Process:               PrusaSlicer [88209]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            PrusaSlicer
Version:               PrusaSlicer PrusaSlicer-2.1.0+-201909160907 (PrusaSlicer-2.1.0+-201909160907)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [88209]
User ID:               501

Date/Time:             2019-09-28 10:26:47.604 +0800
OS Version:            Mac OS X 10.14.6 (18G87)
Report Version:        12
Anonymous UUID:        657A0C7D-19B7-8CFE-15C9-878EA835EDBA

Sleep/Wake UUID:       45D9406E-F898-412F-A0DC-DA230A8CF135

Time Awake Since Boot: 770000 seconds
Time Since Wake:       8000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [88209]

VM Regions Near 0:
--> 
    __TEXT                 0000000104869000-000000010578f000 [ 15.1M] r-x/rwx SM=COW  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.prusa3d.slic3r/           	0x0000000104e169d2 wxEventFunctorFunctor<wxEventTypeTag<wxDataViewEvent>, Slic3r::GUI::ObjectList::ObjectList(wxWindow*)::$_0>::operator()(wxEvtHandler*, wxEvent&) + 82
1   com.prusa3d.slic3r/           	0x0000000105234742 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 210
2   com.prusa3d.slic3r/           	0x000000010523449b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
3   com.prusa3d.slic3r/           	0x00000001052343b7 wxEvtHandler::ProcessEvent(wxEvent&) + 103
4   com.prusa3d.slic3r/           	0x0000000104fd3887 -[wxCocoaOutlineView outlineViewSelectionDidChange:] + 103
5   com.apple.CoreFoundation      	0x00007fff3ebe4b96 __CFNOTIFICATIONCENTER_IS_CALLING_OUT_TO_AN_OBSERVER__ + 12
6   com.apple.CoreFoundation      	0x00007fff3ebe4b10 ___CFXRegistrationPost_block_invoke + 63
7   com.apple.CoreFoundation      	0x00007fff3ebe4a7a _CFXRegistrationPost + 404
8   com.apple.CoreFoundation      	0x00007fff3ebecf28 ___CFXNotificationPost_block_invoke + 87
9   com.apple.CoreFoundation      	0x00007fff3eb558b4 -[_CFXNotificationRegistrar find:object:observer:enumerator:] + 1642
10  com.apple.CoreFoundation      	0x00007fff3eb54c67 _CFXNotificationPost + 732
11  com.apple.Foundation          	0x00007fff40ddaf5b -[NSNotificationCenter postNotificationName:object:userInfo:] + 66
12  com.apple.AppKit              	0x00007fff3c32e942 -[NSTableView _sendSelectionChangedNotificationForRows:columns:] + 228
13  com.apple.AppKit              	0x00007fff3c527a81 -[NSTableView mouseDown:] + 6161
14  com.apple.AppKit              	0x00007fff3c526061 -[NSOutlineView mouseDown:] + 73
15  com.apple.AppKit              	0x00007fff3c3ad937 -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:] + 5724
16  com.apple.AppKit              	0x00007fff3c2e41a6 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 2295
17  com.apple.AppKit              	0x00007fff3c2e3667 -[NSWindow(NSEventRouting) sendEvent:] + 478
18  com.prusa3d.slic3r/           	0x0000000105066435 -[wxNSWindow sendEvent:] + 117
19  com.apple.AppKit              	0x00007fff3c182e6b -[NSApplication(NSEvent) sendEvent:] + 331
20  com.prusa3d.slic3r/           	0x0000000104fe4d28 -[wxNSApplication sendEvent:] + 136
21  com.apple.AppKit              	0x00007fff3c1715e0 -[NSApplication run] + 755
22  com.prusa3d.slic3r/           	0x000000010505ca4e wxGUIEventLoop::OSXDoRun() + 110
23  com.prusa3d.slic3r/           	0x0000000105220224 wxCFEventLoop::DoRun() + 52
24  com.prusa3d.slic3r/           	0x00000001051a6f58 wxEventLoopBase::Run() + 88
25  com.prusa3d.slic3r/           	0x000000010517b650 wxAppConsoleBase::MainLoop() + 128
26  com.prusa3d.slic3r/           	0x000000010502176a wxApp::OnRun() + 26
27  com.prusa3d.slic3r/           	0x00000001051d3d61 wxEntry(int&, wchar_t**) + 145
28  com.prusa3d.slic3r/           	0x000000010487314f Slic3r::CLI::run(int, char**) + 11919
29  com.prusa3d.slic3r/           	0x0000000104875d4d main + 429
[... 450 lines cut ...]
VM_ALLOCATE                        856K       29 
__DATA                            44.1M      367 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       229.5M        7 
__TEXT                           305.8M      370 
__UNICODE                          564K        1 
mapped file                       72.1M       51 
shared memory                     3124K       14 
===========                     =======  ======= 
TOTAL                              1.4G     1178 
TOTAL, minus reserved VM space     1.0G     1178 

Model: MacBookPro12,1, BootROM 186.0.0.0.0, 2 processors, Intel Core i5, 2,9 GHz, 8 GB, SMC 2.28f7
Graphics: kHW_IntelIris6100Item, Intel Iris Graphics 6100, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1867 MHz, 0x02FE, 0x4544464132333241324D412D4A442D460000
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1867 MHz, 0x02FE, 0x4544464132333241324D412D4A442D460000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.77.61.2 AirPortDriverBrcmNIC-1305.8)
Bluetooth: Version 6.0.14d3, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SD0256F, 251 GB
USB Device: USB 3.0 Bus
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
```

lukasmatena, 2019-09-28:
I reproduced the Linux crash (with the right button) and tried to fix it with https://github.com/prusa3d/PrusaSlicer/commit/5ded376fe2a0ef897e9ee6b8d2f8f82dca17848c.

@YuSanka Could you please review my fix and finish whatever is needed to make it work on all the other platforms?

By the way, this bug was in PrusaSlicer2.0.0 already, it is not broken recently.

YuSanka, 2019-09-30:
> @YuSanka Could you please review my fix and finish whatever is needed to make it work on all the other platforms?
I tested the fix on other platforms. Everything is fine.

@anagingcompiler, thanks for your report. This fix will be a part of next release.
@lukasmatena, thanks for a fix ;)
Close.

commit 5ded376f (2019-09-28): An attempt to fix #2997
More fixes might be needed on the other platforms
closed 2019-09-30 by YuSanka