prusa3d/PrusaSlicer #2998: Extra perimeter created out of the part size

gregsaun, 2019-09-27:
### Version
PrusaSlicer 2.1.0+win64

### Operating system type + version
Windows 10 Pro 64b build 18362

### 3D printer brand / version + firmware version (if known)
Bear Upgrade 2.1-dev from Prusa MK3S

### Behavior
A hole close to the outside perimeters creates an extra perimeters outside the part. This might be linked to the elephant foot compensation.

[image]

[image]

If the hole is closer it could split this extra perimeters is several small perimeters.

The Expected result is to not have perimeters at all as this section is too small to be printed.

#### Project File (.3MF) where problem occurs

[ps210_bug_elephant_01.zip](https://github.com/prusa3d/PrusaSlicer/files/3663887/ps210_bug_elephant_01.zip)

Jebtrix, 2019-09-28:
OK, this one is a a little weird..

* Open attached project, slice, bad result
* Change Printer to default MK2.5S, slice, bad result
* Change Printer to default MK3S, slice, result is good
* Change Printer back to original embedded profile or MK2.5S, they both slice fine now

bubnikv, 2019-10-31:
I am working on an adaptive algorithm, that does not shrink parts narrower than two perimeter lines. This is the preliminary result for your model:

[image]

gregsaun, 2019-10-31:
Thank you very much Vojtech!

So the idea is that elephant foot compensation does not shrink details that are smaller than 2 perimeters width? Why two and not one?

bubnikv, 2020-01-02:
> So the idea is that elephant foot compensation does not shrink details that are smaller than 2 perimeters width? Why two and not one?

1) Two is more conservative than one in regard to bed adhesion.
2) Any slicer I am aware of works primarily with closed curves, thus minimum two lines. Single line walls are usually detected after the closed curves.

The elephant foot compensation improvements are part of PrusaSlicer 2.2.0-alpha2. Closing.

closed 2020-01-02 by bubnikv