prusa3d/PrusaSlicer #2999: The "Show" Dropdown list goes out of the screen

Kachidoki2807, 2019-09-27:
### Version
PrusaSlicer-2.1.0+win64-201909160915

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 34,312MB
OpenGL installation
GL version:   4.5.13447 Compatibility Profile Context 20.19.0.32837
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon R7 200 Series
GLSL version: 4.50
(I didn't copy/pasted the extensions)

**Main monitor 2560x1440
Second monitor 1680x1050**

### Behavior
The "Show" dropdown list with the checkboxes (Plater view after slicing) goes outside the scren while in full screen on my second monitor. Moreover, the focus of the main window is lost, that means that the list is a window itself (thus  the Alt+Print Screen will make a screenshot of this small part only). I have no problem with the first monitor.

_Is this a new feature request?_
No.

I feel that the different vertical resolution between the two monitor is the root cause.

[image]

bubnikv, 2019-09-30:
unfortunately I cannot reproduce it. Do you have auto hiding of the system
bottom tool bar enabled?

Kachidoki2807, 2019-09-30:
Hi @bubnikv ,
No I do not use the auto-hiding taskbar. I just reproduced it on a second computer with 3 monitors. I searched around a bit, but finally found out how to reproduce it:
[image]

The second monitor, the one with the lower resolution (here the right one, number 1) must be placed upper that the main monitor (number 3).

bubnikv, 2019-09-30:
I see, it is likely an issue in wxWidgets or maybe Windows common controls,
that do not detect the boundary of each monitor, but it calculates with the
bounding box of the three monitors.

Kachidoki2807, 2019-09-30:
I doubt that it calculate the bounding box of the three monitors, it looks like it take only the main monitor as reference. In this example, I had the issue on the No1, but not on the No3. However the No3 is upper than the No2.

I have to point out that the "normal" dropdown list works as expected, the issue occurs only with this special dropdown with checkbox that take the focus.

But the most important is that you now know where to look at. ;-)

bubnikv, 2020-10-16:
BTW our tester @FidelCapo reports the same issue on the new OSX Big Sur.

drelich, 2020-10-20:
> BTW our tester @FidelCapo reports the same issue on the new OSX Big Sur.

I can't even get it started on macOS Big Sur latest beta let alone see that issue, haha.

foreachthing, 2020-10-20:
> I feel that the different vertical resolution between the two monitor is the root cause.

No, I got two 4k monitors, both at same resolution, and the same issue.

My guess is, that the screen size is measured without considering the taskbar.

It just hides behind the taskbar if PS is not in fullscreen (and just the right distance to the taskbar):
[image]

bubnikv, 2020-12-11:
Let's see whether https://github.com/prusa3d/wxWidgets/commit/ba9f2331149a4a81cef12348572cfa799c1b411a
will fix the windows issue on Windows.

https://github.com/prusa3d/wxWidgets/commit/b91fcebb2817af2d2df7bb1f146ce2a2bb9ad348 should fix the "below taskbar" issue on Windows.

I hoped https://github.com/prusa3d/wxWidgets/commit/dfa511e5f88a0ea37364339a1cf3dc32ff726ed7 would fix the same issue on OSX, but it did not.

bubnikv, 2021-04-14:
Fixed in PrusaSlicer 2.3.0-rc by fixing wxWidgets.

closed 2021-04-14 by bubnikv

Kachidoki2807, 2021-04-14:
Hi,
Did you meant 2.3.1-rc? I just tried with 2.3.0+win64, the issue still exists.

bubnikv, 2021-04-14:
2.3.1-rc. We have just released it.

Kachidoki2807, 2021-04-14:
Thank you, tested with 2.3.1-rc, issue fixed!
See you.