prusa3d/PrusaSlicer #3000: Interleaved perimeters/infill when modifier is removed

haarp, 2019-09-27:
### Version
PS 2.1.0 final on Linux

When certain modifier features are removed from a piece, the preview shows buggy interleaved perimeters and perimeters. I haven't checked the gcode produced, but suspect it might be broken too.

[image]

1. Grab both STLs from https://www.thingiverse.com/thing:976049/files
2. Open avvain_rev2.stl, add avvain_vahvistus_rev2.stl as modifier mesh
3. Add settings to modifier mesh: Infill: rectilinear 100%, perimeters: 5 (higher than default)
4. Slice
5. Remove perimeter setting from modifier mesh
6. Slice
7. Bug triggered

Thanks!

Jebtrix, 2019-09-27:
I have seen this too. I didn't want to report it until I was able to nail down an exact recipe of actions to duplicate it. Those steps do it with any objects, nice job! I've already it where I could get a modifier to have no perimeters when it should. Once I changed some settings to nudge to have a perimeter I wasn't able to get to have none again. When I tried to reproduce that problem I couldn't figure out what did it. I'm guessing it may share the same cause.

Jebtrix, 2019-09-27:
To make the devs life easier, just slice then remove Perimeter object setting:

[issue_3000_demo_project.zip](https://github.com/prusa3d/PrusaSlicer/files/3664623/issue_3000_demo_project.zip)

bubnikv, 2021-06-03:
I can reproduce, thanks for the data.

closed 2026-09-29 by prusaslicer-bot[bot]