prusa3d/PrusaSlicer #3001: Elephant foot compensation corrupts circles

haarp, 2019-09-27:
PS 2.1.0 final on Linux

Enablig elephant foot compensation tends to corrupt arcs and circles.

[image]

Repro:

1. Take any random part
2. Add cylinder modifier mesh
3. Add settings: No perimeters, no infill, no top/bottom layers
4. Slice -> a clean hole through the part
5. Set elephant foot compensation to 0.2mm
6. Slice -> First layer of hole is corrupted as seen in the screenshot

Thanks!

Kachidoki2807, 2019-09-28:
Maybe it is related to #2998

bubnikv, 2019-10-31:
@haarp Please provide a 3mf so we can retest a fix. Thank you.

haarp, 2019-10-31:
@bubnikv It shouldn't matter as it happens with all parts, but here you go: 
[avvain_rev2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3794361/avvain_rev2.3mf.zip)

bubnikv, 2020-01-02:
The elephant foot compensation improvements are part of PrusaSlicer 2.2.0-alpha2. Closing.

[image]

closed 2020-01-02 by bubnikv