prusa3d/PrusaSlicer #3003: Weird behaviour of the download button on the PrusaSlicer project page

DominikPalo, 2019-09-28:
### Behavior
1. Navigate to the https://www.prusa3d.com/prusaslicer/
2. A button with the text "DOWNLOAD THE LATEST VERSION 2.1.0" is displayed
3. After a while, the text changes to "DOWNLOAD THE LATEST VERSION 2.0.0"
4. Then it reverts back to "DOWNLOAD THE LATEST VERSION 2.1.0" after few seconds (so it is continuously switching between 2.1.0 and 2.0.0)

[image]

Browser: Chrome 77.0.3865.90 (tried also in the incognito mode)

lukasmatena, 2019-09-28:
Thanks, passed to the content team.

lukasmatena, 2019-09-30:
Fixed by the content guys. Closing.

closed 2019-09-30 by lukasmatena