prusa3d/PrusaSlicer #3004: Flickering with Wayland

RyuzakiKK, 2019-09-28:
### Version
Version 2.1.0

Installed from the Arch Linux repository `$ pacman -S prusa-slicer`

### Operating system type + version
Arch Linux

My system info https://gist.github.com/RyuzakiKK/5a844fab6c08cc46bbff4d4ef46fb920

### Behavior
* Open PrusaSlicer in a wayland session
* Move the mouse around in the main plater window
* The whole scene will start to flicker
[image]

This problem seems to be related to wayland because with an X11 session I was unable to reproduce it.

lukasmatena, 2019-09-28:
Can you try an appimage from our github (https://github.com/prusa3d/PrusaSlicer/releases)? According to https://www.archlinux.org/packages/community/x86_64/prusa-slicer/, the Arch package links dynamically to wxWidgets 3.0, which is not entirely supported by PrusaSlicer.

Our builds are statically linked to wxWidgets 3.1, more info in https://github.com/prusa3d/PrusaSlicer/issues/2995.

RyuzakiKK, 2019-09-28:
Sadly I still get this issue also with the AppImage binary (and a Wayland session).

bubnikv, 2019-09-30:
Did you try any other OpenGL based application in your Wayland setup?

RyuzakiKK, 2019-09-30:
@bubnikv yes, for example both games and `glxgears` works as expected.

Am I the only one facing this problem with Wayland? Probably this could be a wxgtk upstream issue.

rtyr, 2021-02-03:
Is this issue still valid? Does it happen with PS2.3.0?

RyuzakiKK, 2021-02-04:
@rtyr With PS 2.3.0 the issue seems to be gone, it doesn't flicker anymore.

closed 2021-02-04 by RyuzakiKK