prusa3d/PrusaSlicer #3005: Unexpected slicing problem

jerryhalstead, 2019-09-28:
### Version
2.1 

### Operating system type + version
MacOS Mojave 10.14.6

### 3D printer brand / version + firmware version (if known)
Prusa MK3s

### Behavior
Imported attached STL and in default layout slicer reports empty layers detected. I orient it so small wing end is on bed and now slicer fills the object with infill and caps the ends. Turn those off and basically nothing inside the model is there. Experimented with thin layer/vertical settings, no difference.

[image]

[image]

 Import same model into Cura or S3D and they slice it just fine and the S3D one printed on MK3s correctly. 

[image]

Original STL and 3mf
[wing.zip](https://github.com/prusa3d/PrusaSlicer/files/3665425/wing.zip)

Jebtrix, 2019-09-28:
If I had to grade how clean the mesh itself is for 3D printing I'd give it an F. 532 non manifold edges. I'm surprised any slicer takes this thing. The wing outer skin itself has no thickness that is why it needs to be vertically orientated. This model to render a picture no problem, but to print yikes. Cura sliced it as completely a solid object for me, I wouldn't call that fine. Looks like simplify3D wins the manifold mesh repair prize here.

jerryhalstead, 2019-09-28:
Wow, that's both funny and sad! I tried running it through prusa's repair but maybe it was too far gone for that. What tool do you use to look at/evaluate the mesh? I never quite know when I get a model in prusa slicer that slices bad if it's a setting I'm missing or the model is just messed up.

Jebtrix, 2019-09-29:
@jerryhalstead I usually look at it in Blender and Meshlab. I was a little harsh. The mesh's main issue is the zero thickness wing skin. Adding an offset to make it solid greatly improves the result since your getting rid of the non manifold feature. If you try to repair this type of non-manifold automatically, it will not magically create a solid skin offset, usually (and what happened here and Cura) is that you end up having faces being created across this shell boundary in the attempt. That makes it manifold, but also turns this sleeve mesh into a solid "capped" object. Tip of the hat to Simplify3D for being able do this one right. 

[image]

Jebtrix, 2019-09-29:
>  I never quite know when I get a model in prusa slicer that slices bad if it's a setting I'm missing or the model is just messed up.

To answer that the best thing you can do is learn some 3D modeling. Blender has a steep learning curve but for this purpose you don't need to try to master texturing, lighting, animating and all the other areas. Just concentrate on clean topology type tutorials. Also read up on what manifold meshes means and non-manifold problem types. In its simplest term, manifold answers the question could this exist in physical world.  If you make a plane (not the flying kind) in a 3D editor, you can see it, move it, rotate it but it has zero width. In the physical world it can't exist. Take a cube, remove the top face. Looks like it could exist, but all sides/faces have zero thickness, it can't. Easiest way to make it manifold, add a face to the top. Now its a solid cube, and would be sliced as such. Once that concept sets in things will start to fall into place for you.

bubnikv, 2019-09-30:
Please check
https://forum.prusaprinters.org/forum/prusaslicer/slicing-rc-plane-wing/

and 

[me109_T_wing_R1 (2).zip](https://github.com/prusa3d/PrusaSlicer/files/3669260/me109_T_wing_R1.2.zip)

jerryhalstead, 2019-09-30:
Thanks everyone. It is just a model I found on thingiverse so I wasn't sure if I needed to mess with advanced prusa slicer settings to get it going, then when I seemed to slice ok in S3D I figured I'd point it out here. Didn't consider that the model was so messed up, but I should have thought of that.

closed 2019-09-30 by jerryhalstead