prusa3d/PrusaSlicer #3006: Belt Printer Support

NAK3DDesigns, 2019-09-28:
### Operating system type + version
_What OS are you using, and state any version #s_ Windows 10

### 3D printer brand / version + firmware version White Knight 1.0
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_  Duet Wifi2 version 2.02

_Is this a new feature request?_ Yes

Yes, please consider adding belt printer support for printers like the White Knight, Powerbelt 3D, Printrbelt, & Blackbelt so that there is an option besides Blackbelt Cura..

Thank you 
Karl Brown
NAK 3D Designs

Rizzist, 2020-10-23:
Any update on this? Would love to see this very soon.

czinehuba, 2020-10-23:
Yes. That would be really great to have belt printer support in Prusa slicer.

eddie-moser, 2020-10-23:
+1 on this support

Polymira, 2020-10-23:
I think that this would be very smart feature add for PrusaSlicer right now. With the 3D Print Mill around the corner for under $1000 USD, I see this printer type gaining in popularity quickly. And I would personally love to see PrusaSlicer at the forefront!

Infinite Z and the option to print X amount of a single part one after the other.

cymonsgames, 2020-10-23:
Please.

SyberxSpace, 2020-10-25:
+1 from me. Whether or not Prusa Research is planning on ever pursuing this tech, adding support for it will make PS a more viable option for other printers. It's been getting easier to add custom machines, but it seems there is a lot of work to do before PS hits Cura levels of universal compatibility.

Allram, 2021-05-01:
This would be a nice feature for Prusaslicer! :)

RopesAreForPussies, 2021-07-08:
+1 From Me! 

Thought I'd give this thread a bit of a bump! I have just got the CR-30 (Creality Print Mill), creality's slicer as well as the others available just arent that great tbh. I would LOVE for support for belt printers to be added, would be even cooler if Prusa added a profile for the cr-30 :)

Maybe they are waiting until they have there own belt printer before adding support to there slicer!

Gspohu, 2021-08-20:
I hope this nice feature will come soon !

Pierro55, 2022-01-14:
+1 from me, the available options right now are a pain to work with

v-chriso, 2022-01-24:
this would be huge

ozzman39, 2022-02-06:
Yea please!!!!

Thesaint7811, 2022-03-09:
Vote from me Too!

suprchunk, 2022-04-24:
Let me also throw my two cents into this and say it would be welcomed and appreciated.

jaymemaurice, 2022-07-26:
I would love to try arachne engine with the belt and have one slicer to rule all the FDM printers. Seems I'm stuck defaulting to Ideamaker.

jmezzacappa, 2022-10-23:
@jaymemaurice 
> I would love to try arachne engine with the belt and have one slicer to rule all the FDM printers. Seems I'm stuck defaulting to Ideamaker.

I found this issue from a Google search (I just ordered an IdeaFormer IR3 V1! :D) and thought I'd add some alternative options here, in case anyone else is searching for belt-enabled slicers...

- Kiri:Moto supports belt printers.
- I _think_ I remember seeing some settings in KISSlicer for "BlackBelt-style" printers.

I'll probably be defaulting to ideaMaker as well. Definitely +1 on wanting to use PrusaSlicer and Arachne with the belt.

cartazio, 2022-12-13:
relevant source code from the older cura fork seems to live here https://github.com/BlackBelt3D/Cura/blob/4e18b197ad7cb929058cacdf7ef8031fb6804bbd/plugins/BlackBeltPlugin/

Riley-au, 2023-02-11:
As a long time Prusa user and a new CR-30 user; it would be great to be able to work with just the prusaslicer.

enzoav, 2023-09-01:
> As a long time Prusa user and a new CR-30 user; it would be great to be able to work with just the prusaslicer.

Agreed. Would really like support for cr30. PrusaSlicer is the best and really bummed I have to use Cura again. 😢

JohnySaak, 2023-10-26:
+1
Its so painful tho slice in other slicers. I use strictly prusaslicer but for my belt printer Im forced to use creality slicer and it just doesn't work.

Pierro55, 2023-10-26:
> +1 Its so painful tho slice in other slicers. I use strictly prusaslicer but for my belt printer Im forced to use creality slicer and it just doesn't work.

I recommend you to try out ideamaker with belt printer support. It is still weird but waaaay better than the cura belt slicer version.

prolwns, 2024-01-19:
+1 please add belt printer support for prusaslicer

vdd9474a, 2024-01-28:
+1  it could be awesome to have a nice and modern slicer for this king of 3D printer and it could be push this slicer as a reference for custom 3D printer

enzoav, 2024-02-14:
> > +1 Its so painful tho slice in other slicers. I use strictly prusaslicer but for my belt printer Im forced to use creality slicer and it just doesn't work.
> 
> I recommend you to try out ideamaker with belt printer support. It is still weird but waaaay better than the cura belt slicer version.

I've been working with IdeaMaker since you recommended it. Still not as good as using PrusaSlicer.  It is better than Cura, but just cumbersome and not as intuitive to organize your own profiles and there are so many pages to flip through to find the setting you're looking for.  Just seems like a program designed by engineers that did not sample it out to non-engineers for notes on UI. PrusaSlicer is so well thought out. I'd go as far to offer to invest in the development of belt printer support @josefprusa 😘 Let me know how much it'll take.

KuroNeko-san, 2024-07-18:
+1 waiting for belt printers in PrusaSlicer, so i can switch from IdeaMaker

Lyude, 2025-01-08:
PrusaSlicer maintainers: if someone magically appeared and offered a PR with support for this, would it be reviewed/accepted? I've been looking at the work for this being done in superslicer and the changes aren't as involved as I thought (at least at a first glance, but I've never worked on slicer code so we'll see!). I can't really guarantee anything since 3D printing is a hobby for me and not a job, but I recently got a baby belt kit for myself and didn't consider the slicer situation - and would love to possibly add support for this if prusa's willing to take it.

jaymemaurice, 2025-01-09:
I'll commit time to testing and end user documentation if you do

enzoav, 2025-01-09:
I can help with testing and anything else I can try to help on. If someone showed me a how to on coding for slicers, I’m sure I could work on that part too. 😋

martinjohuber, 2025-03-02:
there is a new campaign for support in Orca Slicer, the code would make it much easier to also implement this in Prusa Slicer: https://www.gofundme.com/f/get-3d-belt-printers-into-orca-slicer

enzoav, 2025-03-15:
> there is a new campaign for support in Orca Slicer, the code would make it much easier to also implement this in Prusa Slicer: https://www.gofundme.com/f/get-3d-belt-printers-into-orca-slicer

I just became the largest donor. 🤓 Let me know when you need beta testers, I'm here to do my part.

KuroNeko-san, 2026-09-02:
Still waiting for implementation. Or developers comment at least...

cjshearer, 2026-09-02:
Unless you're dead set on this being in PrusaSclicer, I would check out [HarrierPigeon's work on OrcaSlicer](https://github.com/OrcaSlicer/OrcaSlicer/tree/belt-printer) ([pinned issue](https://github.com/OrcaSlicer/OrcaSlicer/issues/14386)).

rcmaniac25, 2026-09-02:
Would still be nice to support in more then one slicer. (and seeing if the bot tries to again be last post so it can close things even after people reply after it)

NAK3DDesigns, 2026-09-02:
Yea, desire to have belt support in the best slicer hasn’t changed.  Tell Joe to stop hating on belt printers and embrace the “Infinite” possibilities.

HarrierPigeon, 2026-09-04:
I don't think anybody hates belt printers, there's just all of like six models of them to have hit the market approximately ever...

now that doesn't mean that they aren't rather fun, and that hasn't stopped me from doing it in Orca yet...

joe-pacheco, 2026-09-04:
I don't know where somebody got the idea that I hate belt printers, but I checked out your version of orca last night, and i'm really excited about it. Thank you for all the work you do

SpeakingOfBrad, 2026-09-04:
Has anybody looked into the possibility of creating a belt slicing plugin for PrusaSlicer 3.0?

HarrierPigeon, 2026-09-04:
I've done an audit, the type of stuff the plugin system is good for is basically exactly not what I need, I could do the simpler half of the pipeline (going from cartesian coordinates to belt printer standard coordinates) but the more complex part, slicing at an angle to the bed, remains slicer engine heavy work

lukasmatena, 2026-09-05:
This is a feature request. And frankly, it is quite unlikely to be implemented.