prusa3d/PrusaSlicer #3007: « Thin walls detection » option deteriorates sharply  the result

OUPS65, 2019-09-28:
Operating system : WIN 10 64bits family, version 1809
3D PRINTER : MK3S FW3.8.0, nozzel 0.25 E3D-ONLINE – PLA 
PrusaSlicer 2.1.0
Is this a new feature request?  I think, Yes    (for me)

[N_MAT_5feux_FO005mm_VN_BV_190913_12c (repaired).zip](https://github.com/prusa3d/PrusaSlicer/files/3665675/N_MAT_5feux_FO005mm_VN_BV_190913_12c.repaired.zip)
[config.zip](https://github.com/prusa3d/PrusaSlicer/files/3665679/config.zip)

He everyboby,
I print some very tiny piece ( railway 1/160 signal). 
Coming from Cura 4.2.1 to your excellent PrusaSlicer 2.1.0.   ( Still learning )
I have a problem with "detect thin wall" option results as I need it to not loose some details.

A: Without the « Detect Thin Walls » option, I get an excellent result. 
Nb : external diameter : 1,6mm, internal diameter : 0.8mm

[image]

B : With this option checked, the result  deteriorates sharply  see  picture.

[image]

The problem is that WO this option, I lose some tiny details wich are no more printed.
C : WITH
[image]

D : WO
[image]

thanks
Philippe

Jebtrix, 2019-09-28:
Thin wall detection is known to be a double edged sword. You should exhaust all other tweak methods first. You can also use modifiers to strategically enable thin wall detection at the specific problem areas/layers instead of entire object to workaround this. That is what I did here with a Height Range modifier:

[image]

[N_MAT_5feux_FO005mm_VN_BV_190913_12c (repaired)_tweaked.zip](https://github.com/prusa3d/PrusaSlicer/files/3665914/N_MAT_5feux_FO005mm_VN_BV_190913_12c.repaired._tweaked.zip)

bubnikv, 2019-09-30:
I plan to implement post processing of touching / overlapping perimeters, so that the perimeters will get thinned or one would trim the other. That should lead to more predictable and stable solution. Unfortunately I cannot say when this will be implemented, there is so much to do.

OUPS65, 2019-09-30:
So many tks.
The positive evolution of prusaSlicer is already so great. Last version generates a slightly clearer prints than C.  Congratulations on the job very well done by the team.
Hope that your improvement will come soon. 
In the meantime, I will use your trick to work around the problem.
kind regards.
Philippe

closed 2026-09-29 by prusaslicer-bot[bot]