prusa3d/PrusaSlicer #3008: Platter Grid dissapears after a few seconds once program is opened

mletchworth85, 2019-09-28:
PrusaSlicer
Version:   2.1.0+win64
Build:     PrusaSlicer-2.1.0+win64-201909160915

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 7 (build 7601, Service Pack 1), 64-bit edition
Total RAM size [MB]: 6,343MB
OpenGL installation
GL version:   3.1.0 - Build 9.17.10.4229
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 3000
GLSL version: 1.40 - Intel Build 9.17.10.4229
Installed extensions:

GL_3DFX_texture_compression_FXT1
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_copy_buffer
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_map_buffer_range
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_timer_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_compiled_vertex_array
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_gpu_program_parameters
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_s3tc
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_transform_feedback
GL_IBM_texture_mirrored_repeat
GL_INTEL_map_texture
GL_INTEL_performance_queries
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_WIN_swap_hint

### Behavior

When opening PrusaSlicer the grid shows on the platter for only a few seconds, then disappears and is blank. This happens every time I run the program.
[image]

Jebtrix, 2019-09-28:
Integrated Intel graphics strikes again. I see your running latest driver version so guess it rules that out. I'm guessing v2.0.0 worked fine?

mletchworth85, 2019-09-28:
Correct, I had no issues with 2.0.0

bubnikv, 2019-09-30:
2.1.0 implements texture compression for reduced graphics memory usage.
2.1.0 shall behave better than 2.0. We cannot rule out a bug on our side,
but until we collect enough data, we cannot do much. For now, you may try
to update your graphics driver.

kgolger, 2019-10-05:
Update graphics driver is not possible. Intel closed support 2016.
I have had the same problem on my Dell Latitude E5420 with HD3000 graphics and W7 Pro.
My solution: I installed Linux Ubuntu 18.04 on the same machine and Slic3r 2.1.0 is running without any problems. So i'm now working with Linux in all 3d-printing matters, but i must say, that it's not really a solution to must change the system or hardware after an update. Graphics memory today is not a problem anymore. I'm working with Inventor 2019 CAD on this machine and have no problems with graphics memory.
"We cannot do much" is is just not enough.

Jebtrix, 2019-10-05:
@kgolger You did see a code commit was added for this problem right?? You won't see the fix unless you compile from git or wait for next alpha.

enricoturri1966, 2019-10-07:
@mletchworth85
Could you please check if this build solves the issue with your graphic card ?
https://www.dropbox.com/s/t1ixsqtatt30pzj/PrusaSlicer-2.1.0%2B239-win64-g5bf3d5aab-201910070704.zip?dl=0

kgolger, 2019-10-07:
Uninstalled current version 2.1.0, installed VC_redist_x64 (VC++), systemrestart, tryed as user and admin, always the same issue, after one second slicer lost the texture mk3.svg on display.

 

https://www.dropbox.com/s/t1ixsqtatt30pzj/PrusaSlicer-2.1.0%2B239-win64-g5bf3d5aab-201910070704.zip?dl=0

enricoturri1966, 2019-10-10:
@mletchworth85
I made another commit which attempts to fix the issue with your graphic card. Could you please check if this build works ?
https://www.dropbox.com/s/fyc2m39bflczfoq/PrusaSlicer-2.1.0%2B288-win64-ga65b8f462-201910100740.zip?dl=0

kgolger, 2019-10-10:
Hello,
 
whatever you changed, this build is working proper with my HD3000 and OpenGL 3.1!!!
 
Thanks a lot!

 

https://www.dropbox.com/s/fyc2m39bflczfoq/PrusaSlicer-2.1.0%2B288-win64-ga65b8f462-201910100740.zip?dl=0

kgolger, 2019-12-16:
Upgraded to Slicer 2.1.1 x64. Same error again. Again no texture to be seen with HD3000. Latest Drivers are installed (OpenGL 3.1 - higher is not possible because not supported from Intel).
A troublesome story with this graphics.....

enricoturri1966, 2019-12-17:
This is because the fix made with a65b8f462dd46c8a5f1410504521f26621aeb882 has not been ported to stable version. It will be part of 2.2.0 release.

bubnikv, 2019-12-17:
PrusaSlicer 2.2 will allocate at most 1/2 of the maximum texture size reported by the OpenGL driver. Such trick seems to fix this issue.
Closing.

closed 2019-12-17 by bubnikv

kgolger, 2019-12-17:
Thanks for fast reply. The only mistake is to work with Intel graphiccards.
Then let's wait for 2.2.0. Will see...

ercoleg, 2020-02-24:
Hello,
I have got a similar issue with prusaslicer 2.2.0 beta 
Is it possible to make it working with an  Intel(R) HD Graphics 3000?

this is the opengl extension viewer report:

Renderer: Intel(R) HD Graphics 3000
Vendor: Intel
Memory: 1024 MB
Version: 3.1.0 - Build 9.17.10.4459
Shading language version: 1.40 - Intel Build 9.17.10.4459

Max texture size: 8192 x 8192
Max texture coordinates: 8
Max vertex texture image units: 16
Max texture image units: 16
Max geometry texture units: 0
Max anisotropic filtering value: 16
Max viewport size: 8192 x 8192
Max uniform vertex components: 512
Max uniform fragment components: 512
Max geometry uniform components: 0
Max varying floats: 41
Max samples: 4
Max draw buffers: 8

Extensions: 129

GL_3DFX_texture_compression_FXT1
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_copy_buffer
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_instanced_arrays
GL_ARB_map_buffer_range
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_shader_bit_encoding
GL_ARB_shader_objects
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
[... 189 lines cut ...]
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI1uiEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI1uivEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI2iEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI2ivEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI2uiEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI2uivEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI3iEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI3ivEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI3uiEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI3uivEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI4bvEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI4iEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI4ivEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI4svEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI4ubvEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI4uiEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI4uivEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribI4usvEXT 
GL_EXT_gpu_shader4 was not found, but has the entry point glVertexAttribIPointerEXT 
GL_EXT_provoking_vertex was not found, but has the entry point glProvokingVertexEXT 
GL_EXT_subtexture was not found, but has the entry point glTexSubImage1DEXT 
GL_EXT_subtexture was not found, but has the entry point glTexSubImage2DEXT 
GL_EXT_subtexture was not found, but has the entry point glTexSubImage3DEXT 
GL_NV_transform_feedback was not found, but has the entry point glGetIntegerIndexedvEXT 
GL_NV_transform_feedback was not found, but has the entry point glGetBooleanIndexedvEXT