prusa3d/PrusaSlicer #3009: Slicer window is slow to move around

I-like-stuff, 2019-09-29:
### Version
2.1.0

### Operating system type + version
Windows 10 Home (Insider)
1903
18990.1

### Behavior
After opening PrusaSlicer and I try to move the window around it is incredibly slow to follow my cursor. I think this started after I installed the first version of PrusaSlicer.

I have included a screen recording, although it introduces some slowness most of what you see is actually happening.

[PrusaSlicerSlow.zip](https://github.com/prusa3d/PrusaSlicer/files/3666458/PrusaSlicerSlow.zip)

salem83, 2020-07-28:
Same here ... 

*Version*
PrusaSlicer-2.2.0+win64-202003211145

*OS*
Name: Microsoft Windows 10 Pro x64
Version: 10.0.19041.388

madow64, 2020-08-18:
I have same problem 
Version
PrusaSlicer-2.2.0+win64
Microsoft Windows 10 Pro x64
Is there a fix for this ?

JarTheHead, 2020-09-01:
Same problem

Version
PrusaSlicer-2.2.0+win64-202003211145

OS: Microsoft Windows 10 Pro x64
Version: 2004
build: 19041.450

mavermc, 2020-09-12:
I got this same problem after I updated to Win 10 to 2004 I'm guessing that's related.

ilimo, 2020-09-21:
Same here win10 2004 - is there a fix yet?

foreachthing, 2020-09-21:
Uhm ... same Windows version, same PS version - no problem.
[image]
Anything I could provide to help to solve this issue?

bubnikv, 2020-09-21:
@YuSanka is working on a cure. We are quite positive we will fix it in the
next PrusaSlicer release. As a side effect, PrusaSlicer will start at least
2x faster.
The issue was the number of Win32 controls we are initiating at the start
of the application. We will instantiate only those Win32 controls that are
shown at the active parameters page in the next release.

bubnikv, 2020-10-17:
It should be fixed in PrusaSlicer 2.3.0-alpha1.

closed 2020-10-17 by bubnikv

trollhunterD, 2020-10-25:
its still in PrusaSlicer 2.3.0-alpha1, the there is less lag when moving the windows but it not as bad as 2.2.0