prusa3d/PrusaSlicer #3012: In multi material setup, modifier not taking effect

jimmy00784, 2019-09-30:
### Version
Prusa Slicer 2.1.0

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
Custom multi material printer. Klipper.

### Behavior
When multiple modifiers are applied to a part (extruder 1) where an extruder change is applied to the first modifier (to 2) and the second modifier changes the extruder back to the original (to 1), the second modifier extruder change (to 1) doesn't take effect.  (Rook1.3mf)
If the second modifier changes the extruder to some other extruder (to 3), the modifier takes effect.  
(Rook2.3mf)

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[Rook1.zip](https://github.com/prusa3d/PrusaSlicer/files/3668643/Rook1.zip)
[Rook2.zip](https://github.com/prusa3d/PrusaSlicer/files/3668644/Rook2.zip)

jimmy00784, 2019-09-30:
Rook1 Editor View - Part ext 1, Modifier ext 2, Modifier ext 1
[image]

Rook1 Preview
[image]

Rook2 Editor View - Part ext 1, Modifier ext 2, Modifier ext 3
[image]

Rook2 Preview
[image]

bubnikv, 2019-09-30:
Isn't it a duplicate of  https://github.com/prusa3d/PrusaSlicer/issues/2657
 ?

jimmy00784, 2019-09-30:
@bubnikv it might be related but definitely not a duplicate. I described a situation where it works the way it is supposed (Rook2) and another where it doesn't (Rook1).
If a second modifier in an overlapping region wants to reset the extruder back to original value, slicer ignores it. But if it is changed to any other extruder value, it works just fine.

bubnikv, 2021-06-03:
[image]
[image]

I suppose this is the effect you wanted to achieve.

In PrusaSlicer 2.4.0-alpha1 the modifiers were reworked to be applied in the order they are presented at the side panel.

closed 2021-06-03 by bubnikv