prusa3d/PrusaSlicer #3013: Only 1st bottom-layer respects infill-settings

WereCatf, 2019-09-30:
Didn't test with other versions of Slicer, but with 2.0.0 and 2.1.0 at least, Slicer seems to only respect the bottom-layer infill-settings for the 1st layer. If you've set Slicer to generate 3 bottom-layers, the two others are always rectilinear, no matter what you change in the settings.

The same thing seems to apply to top-layers as well, but in reverse: only the topmost-layer respects infill-settings, the others are always rectilinear.

JohnnyDeer, 2019-09-30:
This is default settings of PrusaSlicer. Why you consider this as a problem? It´s fastest way to fill and this layers are not visible on model.

WereCatf, 2019-09-30:
> 
> 
> and this layers are not visible on model.

Depends on the filament's translucency.

JohnnyDeer, 2019-09-30:
Yes, that´s true. What model do you want to print? It´s needed to use different infill?

Jebtrix, 2019-09-30:
This should also be possible with modifiers IF we had the option to set Top/Bottom infill patterns to solid manually. For example this is as close as it gets for Concentric since the center area gets no infill like Top solid version:
[image]

rtyr, 2021-02-03:
Solid infill layers are printed with rectilinear infill. Most of the infill patterns are not suitable for 100% density. There is similar issue opened https://github.com/prusa3d/PrusaSlicer/issues/5803. I will close this issue, the other one has more suiting title and description.

closed 2021-02-03 by rtyr