prusa3d/PrusaSlicer #3014: Max print speed setting has no effect

rustybox, 2019-09-30:
### Version
2.1.0+linux64 appimmage

### Operating system type + version
Debian 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S 

### Behavior

Open print settings
Choose built in profile e.g 0.2mm speed (max speed 200mm/s)
Slice and note estimate

Edit profile to have max speed e.g 5mm/s
Slice and note estimate same as original slice

_Is this a new feature request?_
Don't think so

Jebtrix, 2019-09-30:
Read the tooltip of Max speed. It is for calculating speeds set to 0 aka AutoSpeed. Your default MK3S profile uses specifically tuned values from Prusa. Autospeed has no influence.

Jebtrix, 2019-09-30:
Read Max print speed tooltip. Its for calculating Autospeed, which is when speed values are set to 0. Default MK3S profile has specifically tuned values set by Prusa. Autospeed is not used so Max print speed does nothing.

rustybox, 2019-09-30:
Fair enough, is there a way to set a global max speed for moves whilst printing like in cura?

Jebtrix, 2019-09-30:
> Fair enough, is there a way to set a global max speed for moves whilst printing like in cura?

If you just wanted set a Max speed cap, technically sounds like something that could be achieved by setting Printer Settings > Machine Limits (not that I recommend doing it that way). Curiously though Print Setting speed values seem to override Printer > Machine Limits feedrates. @bubnikv Is that intended?

> Is the intent to set all to 0 and then set the max speed?

I would not use Autospeed. Read this for some insight #2986. If you have a genuine Prusa, those values have been carefully curated to work best. That tuning is part of what you pay for. Setting all autospeed will be not be most optimal for finish or print time. I think of it more as a function for someone that has a printer with no profile available from the manufacturer, too inexperienced (lazy?) to pick/find optimal values, or possibly testing/debugging.

> A global override would be nice

Do you mean capping max speed or global percentage modifier of all speeds? A global percentage could be useful and I believe has been asked for already.

rustybox, 2019-09-30:
My enquiry is around absolute mm/s limiting, purely based on descriptions from some items on thingiverse which essentially write vertical text in overhang and suggest 5mm/s, I assume setting external permiter to 5mm/s would be just fine.

Perhaps I don't even really need it.

In cura it seems to just be a shortcut to set a variety of other config options in a single place.
[image]

Percentage is pretty much supported out of the box on my printer in the tune menu but presumably would be dead easy to add in slicer as it's just a M220 for the entire print.

Jebtrix, 2019-09-30:
Ah, a straight up cap. You could set all to autospeed then use 5 as max for that, yes.

Jebtrix, 2019-09-30:
Though the external perimeter setting seems more sensible to me.

rustybox, 2019-09-30:
Would there be any interest in adding it as a feature as a combined option? 

Might make for an easy pull request to get my feet wet in the codebase.

rtyr, 2021-02-03:
This setting is meant for autospeed feature as mentioned above. You can use maximum volumetric speed for similar purpose. Closing.

closed 2021-02-03 by rtyr