prusa3d/PrusaSlicer #3015: Default print configurations disappear when custom configurations are used

gmcentyre, 2019-09-30:
PrusaSlicer 2.1.0-Win64

Windows 10 Pro - 64 bit operating system

Prusa MK3S , upgraded from original MK3 purchase 6 months ago.  Only modifications after upgrade were nozzle size adjustments which led to this issue.

Problem started when I created a new profile after switching to a .5mm nozzle.  I adjusted all the settings where I wanted them, and then saved a new configuration with a new name.  At that point all the default configurations disappeared and the only profile now present is the new one I created.  Obviously I don't want to lose the default configurations when creating new ones, hopefully this is a simple fix.

[image]

Jebtrix, 2019-09-30:
When you created new Printer profile for 0.5mm nozzle did you create it from scratch? If you don't have this in Printer Settings > Notes it will do that:

```
Don't remove the following keywords! These keywords are used in the "compatible printer" condition of the print and filament profiles to link the particular print and filament profiles to this printer profile.
PRINTER_VENDOR_PRUSA3D
PRINTER_MODEL_MK3
```

Another place to check is Print Settings > Dependencies, by default Compatible printers is set to All. The Compatible printers condition by default is:

`printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK3.*/ and nozzle_diameter[0]==0.4 and ! single_extruder_multi_material`

You will need to change the `nozzle_diameter[0]==0.4` to `nozzle_diameter[0]==0.5`

Jebtrix, 2019-09-30:
If you pick a different Printer with a different nozzle diameter, Print Settings that don't match that nozzle will be filtered. Changing that behavior is difficult because you can't change built-in profiles to remove the `and nozzle_diameter[0]==0.4` compatible check to have it show up in both.

gmcentyre, 2019-09-30:
Thanks for the replies Jebtrix.  I started with the standard 0.6mm nozzle configuration and edited the print settings as I needed.  However, I never tried closing and reopening the software.  On reopening this afternoon the original configurations were back in the list along with my new configuration.  Here's proof that the age old #1 question from the troubleshooting guide..  "Did you power cycle it?"

[image]

closed 2019-09-30 by gmcentyre