prusa3d/PrusaSlicer #3016: Copies do not print in order added

GaryAitken, 2019-09-30:
### Version
2.1.0+win64

### Operating system type + version
Win7Pro SP-1

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
Using the copy button (Plus Icon) to add additional copies of an stl file, then manually arranging, does not produce the desired print order.

Instead of copying, re-importing the same stl file and manually arranging *does* produce the desired print order.

To reproduce:
  Import an stl file, then successively make copies and place as follows:
     4     5    6
     stl    2    3
  slice and expord g-code
Expected Results:
    Should print in the order indicated: stl, 2, 3, 4, 5, 6
Actual Results:
   Prints in order: stl, 4,  5,  6,  3,  2
Is this a new feature request?  No

Related issues:
#810: Sequential printing: Order of objects not known
#486: Printing in Manual Order
#1338: Sequential Printing Priority menu item is missing
#1662: Complete Individual Objects should print from front to back
# 1052 Complete individual objects + increase copies + arrange = random print arrangement and order printed

JohnnyDeer, 2019-10-01:
Hi, thank you for summing this issue. It´s actual on our To Do list, hopefullly we´ll solve it soon.

GaryAitken, 2019-10-02:
One additional note.  While adding an object using the big '+' button in the center of the top menu bar doesn't preserve ordering, adding an object using the "Add... (Ctrl-I)" (Cube with '+' icon) button on the far left *does* work.  It opens the add object dialog directly, where you can choose the .stl file again.  It's a bit more convenient than file/import.

bubnikv, 2020-02-04:
The upcoming PrusaSlicer 2.2.0-alpha4 will print sequentially in the order displayed at the object list. Also the object list will support drag & drop to reorder.
Closing.

closed 2020-02-04 by bubnikv