prusa3d/PrusaSlicer #3018: PrusaSlicer support contact Z distance setting is being reset.

AmazeCPK, 2019-09-30:
### Version
2.1.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S FIRMWARE 3.8.0

### Behavior
Prusaslicer's "Print settings" default the support Z contact distance to 0 (soluble).
This occurs after the model has been sliced, and sent off to octoprint with the "Start printing after upload" checkbox ticked in Prusaslicer.

Steps:

- Adjust print settings, and save.
- Slice Model
- Click "Send G-code" button
- Have "Start printing after upload" box checked
- Hit OK
- 
- Go back to print settings menu
- Go back to Support material.
- Contact Z distance has now defaulted back to 0 (soluble)

This bug is intermittent, and not guaranteed to happen 100% of the time, This has happened numerous times however. I have not reproduced the bug any other way than sending the g-code to the printer.

The software does not need to be closed to reproduce this bug.
This bug also seems to occur most often when the software is left open for an extended period.

rtyr, 2021-02-03:
Is this issue still valid? Does it happen with 2.3.0?

rtyr, 2021-03-22:
No response. Not reproducible with 2.3.0. Closing.

closed 2021-03-22 by rtyr