prusa3d/PrusaSlicer #3019: Config: Printer Settings/General/Hostname and API key is not saved in project

ochm, 2019-10-01:
### Version
PrusaSlicer 2.1.0+win64

### Behavior
In printer profile is Host and API key saved and can be reused in any project.
But if i save project and open it again, the Host and API fields are empty.

kiklhorn, 2019-10-03:
+1
I like if there will be two options - 
1) Save project
2) Save project with sanitize

3Dietrich, 2019-12-03:
+1 
on macOS 10.12.6
In case you can't reproduce I took a screen capture: [Slicer_API lost on recall.mov.zip](https://github.com/prusa3d/PrusaSlicer/files/3918990/Slicer_API.lost.on.recall.mov.zip)

bubnikv, 2020-03-15:
Is it still an issue with PrusaSlicer 2.2.0-rc3?
We changed the behavior, so that the OctoPrint info is not saved into 3mf due to privacy issues when sharing the file, but when opening the 3mf, the OctoPrint is pulled from the local profile referenced from the 3mf file.

3Dietrich, 2020-03-15:
sorry, I tried one rc and got total chaos and errors also in Silcer 2.1. I repaired it circumstantially and desided to not test 2.2. anymore and wait for the release..

ochm, 2020-03-20:
the problem is not in RC2, probably in RC3 is not too. (i didn't tested it yet).

bubnikv, 2020-03-20:
This works:

1) Define Host IP address and Host API Key
2) Save your profile XXX.
3) Save a 3MF using this profile XXX. The Host IP address and Host API key have to be saved to the local preset file, thus marked as non-modified.
4) 3MF was saved WITHOUT the Host IP address and API key, but referencing the profile XXX which saves the Host IP address and Host API key.
5) Reload the 3MF. The Host IP address and Host API key are missing in the 3MF for privacy reasons, but the 3MF references the profile XXX, which contains this information, therefore the Host IP address and Host API Key is loaded from the profile XXX and everything is rosy.

What is no more supported:
If you change the Host IP address or Host API key and you don't save the Printer profile before storing the 3MF, when reloading the 3MF the changes in the Host IP address or Host API key are lost.

You see, for privacy reasons we made a breaking change in handling the Host information, but I hope this explanation is clear for you to understand what is happening and why, and you will be able to continue to use PrusaSlicer productively.

bubnikv, 2020-10-17:
The privacy reasons explained, closing.

closed 2020-10-17 by bubnikv