prusa3d/PrusaSlicer #3021: Incorrect slice and arrange.

Shama71, 2019-10-02:
### Version
2.1.0+win64
### Operating system type + version
PrusaSlicer
Version:   2.1.0+win64
Build:     PrusaSlicer-2.1.0+win64-201909160915

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18990), 64-bit edition
Total RAM size [MB]: 10,617MB
OpenGL installation
GL version:   4.6.0 NVIDIA 388.73
Vendor:       NVIDIA Corporation
Renderer:     GeForce GT 620M/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_gpu_shader_int64
GL_ARB_half_float_pixel
[... 260 lines cut ...]
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
For all
### Behavior
Incorrect slice and arrange.
[image]
[image]
#### Project File (.3MF) where problem occurs

[e3d v6 - fan nozzle.zip](https://github.com/prusa3d/PrusaSlicer/files/3679426/e3d.v6.-.fan.nozzle.zip)
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-10-02:
I can't explain the arranging bug but the slice problem is because of the mesh. A cross-section of the model explains itself:

[image]

Shama71, 2019-10-03:
Thank.
Corrected the model.

closed 2019-10-03 by Shama71