prusa3d/PrusaSlicer #3023: 'Bad Allocation Error' When slicing

RenkFasze, 2019-10-02:
Version:2.1.0+win32

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

Win 7 Pro Service Pack 1

MK3s  Firmware 3.7.2-2363
No mods, all stock
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

When slicing larger or multiples of files on a plate I get a 'bad allocation' error and a notification that I may be running out of RAM space. 
Added more RAM even though I have never had this issue. Have also watched my systems resource manager and I never reach past 6/24GB RAM that I have available. 

The file used is irrelevant to the error as long as its semi complex or multiples. 

I just noticed that Im running the 32 bit version of the slicer on a 64bit machine. I looked for the 64 bit version of the slicer and I can't find an option for it.

[image]

Thanks

Kachidoki2807, 2019-10-02:
Do you think it is related to #2991?
For the 64bit version of PrusaSlicer 2.1.0 here is a [direct link](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.1.0/PrusaSlicer-2.1.0+win64-201909160915.zip) from the [releases page](https://github.com/prusa3d/PrusaSlicer/releases).

bubnikv, 2019-10-02:
There are hints, that there is an error in the "Prusa3D Windows drivers"
installer, and it installs Win32 PrusaSlicer binary on Windows 7 64bit. We
will verify that.
For now, you can just pull the binary from the github releases page.

lukasmatena, 2020-03-04:
New installer that will be used for 2.2.0 final should recognize the system correctly. Let us know if it doesn't.

Thanks for reporting. Closing.

closed 2020-03-04 by lukasmatena