prusa3d/PrusaSlicer #3024: Start G-code variable issue

halentech, 2019-10-02:
### Version
2.1.0

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
generic printer Zonestar P802QR2

### Behavior
For picking right tool before print (parking unused extruder and pick right extruder) is needed that _current_extruder_ variable contains valid value. In this variable should be first used extruder for printing in current print process. I mean in the start g-code section. In this section current_extruder variable is un-initialized so it is unusable for anything.

If I place new object, and select that will be printed with second extruder, I expecting that in the current_extruder variable will be value: 1. If I run slicer, then this condition variable is totally ignored.

for example start g-code - move to correct location based on which extruder is first one:
{if current_extruder == 0}G1 X1.6 F4800 ;Extruder 0 location
{if current_extruder == 1}G1 X83 F4800 ;Extruder 1 location
{if current_extruder == 2}G1 X181 F4800 ;Extruder 2 location
{if current_extruder == 3}G1 X262 F4800 ;Extruder 3 location

Jebtrix, 2019-10-02:
Use `initial_tool` placeholder variable instead of `current_extruder`.

Jebtrix, 2019-10-02:
Depending on what your needs are there is a **Extruder offset** to handle multiple-extruder printers.

halentech, 2019-10-02:
initial_tool is working, thank you. No wiki informations… :-(

closed 2019-10-02 by halentech

Jebtrix, 2019-10-02:
If you install a default Prusa MMU type printer configuration (ie MK3 MMU2) you can use it's gcode as a general reference.