prusa3d/PrusaSlicer #3026: [Feature Request] bridge infill width

Schild0r, 2019-10-03:
Hey there, I spent a few days now on wondering why I am getting very thin bridge infill:
[image]
turns out I used a wrong printer profile which had the nozzle size of .2 instead of .4 everything else looked normal though.
This makes me wonder why the bridge infill seems to be the only feature that width is calculated over nozzle diameter without an override option.
So it would be nice to be able to not only set the bridge flow ratio but also the bridge infill width.

Jebtrix, 2019-10-05:
Think about what a bridge extrusion is. It supposed to cross air as a round extrusion. To be "perfectly" round it needs to equal the nozzle diameter. It's not physically possible to create a larger extrusion width in mid air. You can however make a smaller width extrusion by lowering flow. This distinction is why extrusion width settings don't logically apply.

Schild0r, 2019-10-06:
Well as long as the first top layer (above infill) is considered a bridge, you don't create it in mid air and thin extrusion lines still cause problems.

Jebtrix, 2019-10-06:
In your picture the extrusion is indeed crossing mid air in between grid infill pattern squares. Its easier to visualize bridging across a bigger distance like a traditional road bridge on two ends but any distance with nothing immediately underneath is considered a bridge extrusion. Of course there is a minimum distance to realistically consider something as bridging.
That being said however the slicer currently does seem to be bridge detection happy with strict rule of "its not solid infill beneath then its using bridge infill". Take this example, a 99% Rectlinear infill cube. No need to realistically bridge infill but its technically not consider solid infill so it does anyway.

[image]

Schild0r, 2019-10-06:
Guess u r right
After all there is no need to make bridging work with a .4mm nozzle on a .2mm nozzle-profile

closed 2019-10-06 by Schild0r