prusa3d/PrusaSlicer #3030: Infill modifier not retained when saving project

ddavidebor, 2019-10-04:
### Version
2.1.0

### Operating system type + version
MacOS Mojave

### Behavior
click on the modifier on the thingy in the middle of the plate.
Set modifier as fill density 100%, accept the switch to linear infill
save project
close prusaslicer
reopen file
the modifier will now be reset at 15% gyroid
Piece broke customer unhappy :(

_Is this a new feature request?_
nope
[pleasedontcopythisiscopyrighted.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3689964/pleasedontcopythisiscopyrighted.3mf.zip)

Jebtrix, 2019-10-04:
Yeah there is a bug here. When you open your project you and click on Infill Object setting you get a info/warning:

> Support Generator
> Supports work better, if the following feature is enabled:
> -Detect bridging perimeters

If you turn Detect bridging perimeters ON saving project will then save the setting as intended.

Jebtrix, 2019-10-04:
Another manifestation of this bug is if you have Detect bridging perimeters ON with 100% Rectlinear as an object setting but then turn Detect bridging perimeters OFF, object setting is automatically reset back to global Fill Density and Fill Pattern value.

JohnnyDeer, 2020-03-12:
Hi, i´m not able to reproduce described situation, even if i use yours 3mf, slicer works how should. Can you please try to reproduce it with 3md provided here? And could you please try, if problem will be in newest slicer version 2.2.0 rc3?

ddavidebor, 2020-03-12:
I do not recall having this issue in the last month, it has probably been fixed already after the save project code was improved.

JohnnyDeer, 2020-03-12:
Then we can consider this issue as solved and close it?

ddavidebor, 2020-03-12:
definetly

closed 2020-03-12 by JohnnyDeer