prusa3d/PrusaSlicer #3031: Cant enter zero in part or modifier location fields

neophyl, 2019-10-04:
### Version
2.1.0 64 bit installed on a Win 7 machine

### Operating system type + version
Windows 7 64bit

### 3D printer brand / version + firmware version (if known)
Not Applicable

### Behavior
Noted by another user on the Prusa forum.  When placing a modifier he cant enter zero in any of the XYZ location boxes to move the modifier around.  I took a look to confirm and also cant.  Same applies to model parts too without a modifier.  You can enter 0.000005 for example and it will replace it with zero and accept that but if you try and manually enter in a zero it reverts to the previous value in there.
That seems like a bug to me.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Any Project, just place a part on the Plater

Jebtrix, 2019-10-04:
Strange it works fine for me.

Jebtrix, 2019-10-04:
Ah, I just realized I have the habit of entering 0 and pressing Enter which works fine. Just changing the value and moving out of textbox does exhibit that problem sporadically.

GWdd, 2019-10-08:
I can recreate the issue - drag the part or set a value to some number like 99, click the value edit box, enter 0, then press ENTER or click another value, and the number reverts to 99. Set 0, click an adjacent box, the 0 remains, but the part doesn't move to zero .. watch video.

[Demo.zip](https://github.com/prusa3d/PrusaSlicer/files/3703394/Demo.zip)

YuSanka, 2019-10-09:
> I can recreate the issue - drag the part or set a value to some number like 99, click the value edit box, enter 0, then press ENTER or click another value, and the number reverts to 99. Set 0, click [...]

Thanks for your report. This bug is known and already fixed. The fix will be a part of next release.

lukasmatena, 2019-11-08:
This is a duplicate of #3108. It should be solved now, the fix will be in PrusaSlicer 2.2.0.

Closing.

closed 2019-11-08 by lukasmatena