prusa3d/PrusaSlicer #3032: Intel HD3000 incompatibility

kgolger, 2019-10-04:
### Version
2.1.0

### Operating system type + version
Windows 7 prof. latest SP

### 3D printer brand / version + firmware version (if known)
MK3S/MMU2S with firmware 3.8.0

### Behavior
Hello community,

my Dell E5420 with i5-2520 CPU and internal graphics HD3000 supports only OpenGL 3.1 which is not compatible with PrusaScli3r 2.1.0 (Always lost textured mk3.svg on screen after one second). The laptop is a bussines model 2014. On earlier Slicer versions there was an option for older graphics cards. Why did Prusa remove this from V 2.1.0. The most of us printer enthusiasts have little budget and do not want to upgrade their computer because of an Slicer-Update. Intel is not interested in an HD3000 update. What does prusa say about that?

Does anyone have a solution about this known problem?

thanks!

Jebtrix, 2019-10-05:
Duplicate of #3008

kgolger, 2019-10-05:
see https://github.com/prusa3d/PrusaSlicer/issues/3008

closed 2019-10-05 by kgolger