prusa3d/PrusaSlicer #3033: Default startup doesn't uses m104 instead of m109 so doesn't wait to start.

tlhintoq, 2019-10-05:
### Version
2.1.0_Win64

### Operating system type + version
Windows 10 pro - 64 bit

### 3D printer brand / version + firmware version (if known)
CR10s-5, Tronxy x4st-400, Ender3pro, Prusa i3mk2

### Behavior
The generated gcode uses m104 (heat but don't wait) to set the temp.

If the user's custom startup gCode includes any kind of priming line (like many use)... The nozzle hasn't come up to temp.

_Then_ after all that, the m109 gets set.  Shouldn't that first m104 on line 18 really be an m109?

```
; generated by PrusaSlicer 2.1.0+win64 on 2019-10-05 at 15:49:12 UTC

; 

; external perimeters extrusion width = 0.68mm
; perimeters extrusion width = 0.68mm
; infill extrusion width = 0.68mm
; solid infill extrusion width = 0.68mm
; top infill extrusion width = 0.60mm
; first layer extrusion width = 0.63mm

M201 X500 Y500 Z500 E10000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1500 R1500 T1500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X40.00 Y40.00 Z0.20 E2.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M104 S250 ; set temperature

; Custom Start G-code
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
G28 ; Home all axes
G92 E0 ; Reset Extruder
G1 F200 E3 ;Prime the extruder with 3mm
G92 E0; zero the extruded length counter

G1 Z2.0 F6000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F1500 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F1500.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G1 Z2.0 F1500 ; Move Z Axis up little to prevent scratching of Heat Bed
M117 Printing...
G5 ; High precision contour
; End of custom start GCode

M109 S250 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
; Filament gcode
```

Jebtrix, 2019-10-05:
Then people would complain it causes oozing prematurely. The reality is you should have M104/M109 in your custom g-code to take control yourself. Then it won't automatically add them in a generic fashion catch all fashion. Look at a default Prusa printer profile for custom g-code reference.

slavedas, 2019-10-07:
Also, if you have a heated bed, I'm pretty sure 2.1.0 will issue the M104, then issue an M190 for the bed. Given the bed normally takes longer to come up to temperature, this means your hot end is normally correct by then.

tlhintoq, 2019-10-07:
@slavedas 
After reporting this I noticed something that I need to investigate to find a pattern, but basically:
I see the M104 on a printer definition I made from scratch.  Just following the prompts: New, custom printer, then defined it.  So it used the M104 instead of M109 based on a new blank default printer definition/template.

I noticed that a printer definition I downloaded from the CHEP (Chuck Hellibee) YouTube channel did *not* behave the same way.  So I'm thinking there is something that can be setup to get this safer behavior, but its not the new printer default.  Generally speaking I like seeing coded defaults/templates to error on the side of safe and workable, not _You need to change these 6 things out-of-the-box_ because a new user doesn't know the 6 things.  But that me.
----
The handling of the bed heater is a whole 'nother issue I was going to investigate further before saying anything but since you brought it up...

I have some printers with built-in heated beds. These tend to be the small Ender3pro, Prusa i3 models.  Others with external Keenovo heaters on their own AC power and control boxes. These are the big CR10s-5 500mm sized machines. Meaning those printers are without heated beds as far as the printer thinks.

So... I slice most things at 0 degree heated bed (no heated bed) and just pre-heat manually.  That way the same gCode file can be put in any printer and treated the same way.  I still add turn heated bed off to the custom ending script because its a no-harm-no-foul safety.

I had a job fail on an Ender3pro that I sliced with PrusaSlicer and realized the bed was cold, so the model let go.
Upon examining the generated gCode I found a change to the heated bed temperature  _from_ zero _to_ zero at line 15,xxx

So apparently the description in the program saying 
> "For printers with no bed heater enter 0"

isn't really accurate.  Code is still generated that tries to manipulate the heated bed.  Setting zero is not acting the same as other programs that have a true boolean toggle for `[x] Heated bed`.

But like I said, I want to experiment with that more before saying it happens *every* time.

Just a thought:   Zero is a valid temperature so using it as a boolean for whether or not there is a heated bed is problematic.  But if you use the coding standard of -1 (not a valid temp) to indicate `no heated bed` you get to keep all your `int` values everyplace in code and still treat it like a special condition.

lukasmatena, 2019-10-07:
@tlhintoq I don't think it is clever to discuss both extruder temperature and bed temperature setting simultaneously. I would start with the original issue. Regarding that, @Jebtrix is absolutely right. This is what the tooltip on the custom start gcode says:

>This start procedure is inserted at the beginning, after bed has reached the target temperature and extruder just started heating, and before extruder has finished heating. If PrusaSlicer detects M104 or M190 in your custom codes, such commands will not be prepended automatically so you're free to customize the order of heating commands and other custom actions. Note that you can use        placeholder variables for all PrusaSlicer settings, so you can put a "M109 S[first_layer_temperature]" command wherever you want.

If you need the extruder ready in the start gcode, heat it up yourself whenever you want, exactly like official profiles for Prusa printers do. Does this solve your problem?

tlhintoq, 2019-10-07:
@lukasmatena 
> @tlhintoq I don't think it is clever to discuss both extruder temperature and bed temperature setting simultaneously.

Well... When they are related and interactive ya kinda have to. Right?

> This is what the tooltip on the custom start gcode says:

Do you really go around every new piece of software and inspect every tooltip on every control as your first order of business?  Yeah me neither.

Does updating your specific user startup gCode work around this problemattic behavior?
Yes.
Of course, if the default behavior was to do what every other slicer does and _by default_ wait for the extruder to get to temp before trying to print - there wouldn't **be** a problem to work around.

Generally speaking, default conditions are what most people want/expect/use.  I think its not unfair to say that most people expect their sliced code to not try to print with a cold extruder - as an out-of-the-box behavior.  And then, if they are not in the 97% - and they have some odd situation where they DO want to print with a cold extruder... like maybe when doing an eSteps calibration where you don't need it even on... Then you can manually set it to m104.

Can you see what I'm getting at?  You expect your car... out of the box... from the factory... not not let you slam into reverse while driving 80mph.  You don't expect that unsafe behavior to be enabled by default and that you have to **set** the control system to block that.

@slavedas 
> if you have a heated bed, I'm pretty sure 2.1.0 will issue the M104, then issue an M190 for the bed. Given the bed normally takes longer to come up to temperature, this means your hot end is normally correct by then.

So if you don't have a heated bed, the printer just takes off with a cold nozzle.  That's the point I'm trying to make. One expects smarter logic than to introduce a race condition where you are relying on the average heating time of the bed to make sure the nozzle is probably hot enough to print.

@lukasmatena 
Do you see now why the bed heater and the extruder heater are in the same conversation?

lukasmatena, 2019-10-09:
> Do you really go around every new piece of software and inspect every tooltip on every control as your first order of business? Yeah me neither.

So in other words, you don't read tooltips, which is why we should change the behaviour to the one you expect. Sorry, I don't see that as a valid reason.

The behaviour you dislike is there on purpose - to provide as much flexibility to the user as possible. One might for example want to do mesh bed levelling before the extruder heats up (to prevent oozing). This would not be possible if the extruder unconditionally heated up before custom start gcode. I didn't make this up, there were discussions along these lines, such as #913. This was added years ago, definitely before https://github.com/slic3r/Slic3r/issues/1726 (same issue as this one).

> You expect your car... out of the box... from the factory... 

You're creating your own custom gcode. That's not a factory state, it requires some knowledge about how to do it.

> So if you don't have a heated bed, the printer just takes off with a cold nozzle.

Can you share a 3MF where the printer starts printing cold (apart from cases with incorrect custom start gcode)? It would be a bug we would like to know about. When there are no temperature changing commands in the custom gcode, you get:

M190 S60 ; set bed temperature and wait for it to be reached
M104 S215 ; set temperature
; custom start gcode
M109 S215 ; set temperature and wait for it to be reached

Which is exactly what the tooltip says and there is no race condition and no cold printing (unless you do it yourselff in the custom gcode), regardless of what the bed temperature is.

> Do you see now why the bed heater and the extruder heater are in the same conversation?

Sadly, I don't. The extruder problem is about understanding how custom gcode works. The bed problem is about whether there should be a concept of a printer with no heated bed (to suppress commands such as M190 S0).

tlhintoq, 2019-10-10:
> So in other words, you don't read tooltips, which is why we should change the behaviour to the one you expect. Sorry, I don't see that as a valid reason.

I didn't say I don't read tool tips.  I said I don't make my first priority to go around and find and ready every single one before doing anything else.  I think I do as most people do: When I think I need more information about blah, I hover on it and read the tooltip.  But sometimes you don't know you need a thing because the software is new to you.  So you don't even know where to look, or what the program metaphores are.  You don't know what you don't know.  ya know?  Tooltips are not meant to replace a manual.

> Can you share a 3MF where the printer starts printing cold (apart from cases with incorrect custom start gcode)? 

I'll see if I still have that file.  Since it didn't work correctly I didn't keep it.
But take a look at the gCode I posted from it.  Whether you want to call the custom part right or wrong can be put aside... That part comes AFTER the default code written by PrusaSlicer.  Meaning, unless one knows to put a hold command in their custom section, it prints without waiting to get to temp.

It is absolutely common to prime the nozzle and draw a priming line before starting the actual object print.  That's why it is so commonly in every other slicer's startup code.

I'm not saying its not easily worked around.  Just add the same M109 to one's custom startup.  All I was saying is that its odd to have to add it to the custom when its the norm 90% of the time and doing the mesh bed level as in your example is not the run-of-the-mill, 10 times a day expectation.

Either way - I'm dropping out of this conversation.  Y'all have your reasons and they are based on user conversations so great.  Its how it works.

I've had to give up on using it for now due to lack of features I use in other softwares.  I'll take a second look at it some time in the future such as release 3.  I think its got a lot of potential.

lukasmatena, 2019-10-10:
> Tooltips are not meant to replace a manual.

That is correct. Lack of proper and up-to-date manual is bad and I understand now what you mean. Manual is currently being written, so hopefully we'll manage to fill that gap soon.

> unless one knows to put a hold command in their custom section, it prints without waiting to get to temp

Yes. That's why one should better know that.

> its odd to have to add it to the custom when its the norm 90% of the time and doing the mesh bed level as in your example is not the run-of-the-mill, 10 times a day expectation.

Well there could be an endless debate between many people here with different opinions on this. Which is why current behaviour was chosen so everyone can customize it the way they want. If we did the change you propose and someone wanted to do the cold mesh bed levelling, then what?

Not to mention that is would be a breaking change, requiring users to change their printer profiles. We are not going to do that, sorry. I agree though, that proper documentation would help.

> I've had to give up on using it for now due to lack of features I use in other softwares

We'll be glad if you share what the missing features are (search the issue tracker first though, there are many open feature requests already).

airscapes, 2019-12-20:
Guys, I am really confused, in the current Slic3r release there are check boxes to wait for bed and wait for Extruder, I can not find them in Prusaslicer 2.1.1 and this thread has not be very helpful. 
How do I get the behavior of the bed heating then the extuder then the custom gcode???
I am coming from S32, spent a few hours with slic3r, exported my profile bundle and imported into PS 2.1.1.  Then looked over all the options.  

**This is the Slic3r code**
; generated by Slic3r 1.3.0 on 2019-12-20 at 15:28:31

; external perimeters extrusion width = 0.50mm (1.96mm^3/s)
; perimeters extrusion width = 0.50mm (3.92mm^3/s)
; infill extrusion width = 0.50mm (3.92mm^3/s)
; solid infill extrusion width = 0.50mm (3.14mm^3/s)
; top infill extrusion width = 0.50mm (2.51mm^3/s)

M107
M190 S60 ; set bed temperature and wait for it to be reached
M109 S215 ; set temperature and wait for it to be reached
M108 S255 ; turn on M2 fans
G28 ; home all axes

**This is the PrusaSlicer code**
; generated by PrusaSlicer 2.1.1+win32 on 2019-12-20 at 20:55:07 UTC

; external perimeters extrusion width = 0.50mm
; perimeters extrusion width = 0.50mm
; infill extrusion width = 0.50mm
; solid infill extrusion width = 0.50mm
; top infill extrusion width = 0.50mm

M107
M190 S60 ; set bed temperature and wait for it to be reached
M104 S215 ; set temperature
M108 S255 ; turn on M2 fans
G28 ; home all axes

How do I fix this as I really want to try out this software .
Thanks!

lukasmatena, 2019-12-20:
Place the M109 in your custom start gcode. That should do it. You can use our stock profiles as an inspiration.

airscapes, 2019-12-20:
So add this to start code at top?
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

ManuelMcLure, 2019-12-21:
If you're putting all those in a row, you don't really need all of the lines. for the first set, this would be sufficient:
```
M104 S[first_layer_temperature] ; set extruder temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
```
(i.e. the ``M140`` right before the ``M190`` is redundant) and for the second it would be sufficient to do
```
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
```
(i.e. both the ``M140`` and ``M104`` are redundant in that sequence). You'd only want to use ``M104`` and ``M140`` if you want to do other stuff while the bed or hotend are heating.

airscapes, 2019-12-21:
Ok, I pulled that first list directly from a Prusa profile and it of course heated both at the same time, which would not work on my printer if I was heating the bed for ABS (takes a long time)   So switching the order worked as expected.  I also thought it strange 2 commands were needed.. Will correct that now.  **Having a 2 configuration check boxes to wait for bed and wait for extruder like all the other software uses would really be appropriate.  Why was it removed from the branch of Slic3r when you guys took it over.  Not checking the boxes would do the same thing as yous does now and not reduce any flexibility that I can see.**

DotNetRob, 2020-04-03:
Has there been any update to this issue?  Just tried out prusaslicer 2.2 on my ender 3 and it started trying to print before the hotend got to temperature.  Took me some digging to find out what was wrong.  Have used cura mainly for over a year and mattercontrol a bit, both use M109 without custom start gcode.

Also is there a list of these replacement variables like [first_layer_temperature] somewhere ?

lukasmatena, 2020-04-04:
> Has there been any update to this issue?

No. People are still expected to have a working printer profile. Where did you get your Ender-3 profile that it does not work? Could you send a 3MF?

> Also is there a list of these replacement variables like [first_layer_temperature] somewhere ?

Just hover over the options in parameter tabs. Other parameters and allowed constructs are documented [here](https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language).

DotNetRob, 2020-04-04:
I created by profile thru the wizard in prusa.

DioGreco, 2020-07-11:
In my view, this is not at all smart. If there is a field to set the temperature of the hotend, even customizing your code, why are you required to add the M109 SXXX in the customization? I don't think this is smart and another, only PrusaSlicer presents this concept. Waiting for an update to fix this.

lukasmatena, 2020-07-12:
> why are you required to add the M109 SXXX in the customization

You are not. If you don't do any custom scripting, the temperature will be set for you, which you could have tested yourself before you asked. As far as I remember, this thread is about making custom gcodes without reading documentation and doing it wrong.

closed 2026-09-29 by prusaslicer-bot[bot]