prusa3d/PrusaSlicer #3034: Edit|Ctrl+Del doesn't do anything (Should be delete all models)

tlhintoq, 2019-10-05:
### Version
2.1.0+Win64

### Operating system type + version
Windows 10 pro 64 bit

Edit menu - Delete all keyboard shortcut ctrl+Del doesn't delete anything.

Jebtrix, 2019-10-05:
Cannot reproduce this on Windows or Linux. Is this for any project or a particular one?

tlhintoq, 2019-10-05:
All projects.  All printer configurations.
Does it only work in a given mode? 
Ah ha... Only works in editor mode (Cntrl+5) not in preview mode (Cntrl+6).  Yet the menu option for the same thing works in both modes

Jebtrix, 2019-10-06:
Ah. Well there is some logic there but if Menu item is available from both mode views then shortcut should work as well.

enricoturri1966, 2020-03-16:
If you guys agree, I would prefer disable this (and any other editing command) while preview is shown

tlhintoq, 2020-03-16:
> 
> 
> If you guys agree, I would prefer disable this (and any other editing command) while preview is shown

Isn't that backwards.  It already works.  So rather than kill a working function how about just fixing the shortcut key for the same menu command?

bubnikv, 2020-03-25:
> Isn't that backwards. It already works. So rather than kill a working function how about just fixing the shortcut key for the same menu command?

I don't find this a good idea. There is a good reason why the toolbar is hidden in print path preview. You should get a visual feedback to your actions, and the visual feedback to these manipulation actions are not there or they are confusing if you are in the print path preview.

IMHO most of the edit menu items should be grayed out if not in Plater. These menu items are currently active even in the preset tabs, while they should be grayed out. I don't think it is a big deal, but we could improve.

[image]

[image]

closed 2026-09-29 by prusaslicer-bot[bot]