prusa3d/PrusaSlicer #3035: different fan speed for outer perimeters

Mike8-I, 2019-10-05:
### Version
_2.01
_About Prusa Slicer_

### Prusa MK3s +MMU

_New feature request_

Stefan of CNC kitchen (youtube channel) showed lately that less cooling improves strength of parts but decreases quality.
At least the option for two different cooling speeds could combine improved strength and good quality. One for the outer perimeter(s) using typically a high value and one for everything else using typically a low cooling speed.

Jebtrix, 2019-10-05:
Duplicate of #2776

closed 2019-10-05 by Mike8-I

jschuh, 2019-10-12:
If you can compile your own build please try PR #2921. I've been running with this patch for several weeks now, and am very happy with it. I don't have a rig to properly test the strength of the resulting parts, but anecdotally I'm now able to print at lower temperatures while getting robust parts with pretty much ideal surface finish. Whereas previously I would I would strengthen parts by making the tradeoff of either reducing the cooling (worsening the surface finish) or cranking up the nozzle temperature (worsening overhangs and bridges).

Mike8-I, 2019-10-16:
Thanks for the precompiled version, very usefull. The fan speed for the outer diameter is working as intended and very useful.Some thoughts: When the fan is running very slow or not running at all for the infill and inner perimeters, it needs significant time to speed up to e.g. 50% rpm. 
It might be useful to give it a "boost" of 100% for a short time , this time depending on the speed difference. E.g.: from 10%to 50% 400ms boost, from 10% to 80% 700ms boost. I understand that this is difficult to implement, because it needs to "chop" the gcode to slices of appropriate length or gather some lines together to the appropriate length.

Verschickt: Sa, 12. Okt 2019 2:21

jschuh, 2019-10-18:
What you're asking for is a global behavior that wouldn't make sense in this patch, and can't really be done by the slicer anway. Issue #327 has a fairly detailed discussion of the problem and why it needs to be handled in the printer firmware rather than the slicer.