prusa3d/PrusaSlicer #3036: Prusa Slicer errors

mcboeing, 2019-10-05:
### Version
2.2.7

### Operating system type + version
Mac OS 10.14.5

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Returns nothing.

### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3 - not modified
Latest firmware

### Behavior
Fusion 360 created model.  Opened in Prusa Slicer works fine.  When model sliced various parts are deleted.  

* _Steps needed to reproduce the problem_
Reproduced every time.

  * _If this is a command-line slicing issue, include the options used_
Not command line.
[image]
[image]

* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-10-05:
You need to attach a zipped .3MF project file to provide any help here.

mcboeing, 2019-10-05:
[Drill jig v4.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3693579/Drill.jig.v4.3mf.zip)

JohnOCFII, 2019-10-05:
Running the model through NetFabb allowed it to be sliced properly in PrusaSlicer.

I'm not sure _what_ exactly is wrong, or why the software didn't flag it as a bad STL.

[Drill jig v4_fixed.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3693608/Drill.jig.v4_fixed.3mf.zip)

Jebtrix, 2019-10-05:
This model has mesh issues. If you right click on the model and split to parts you can see the mesh pieces (it is not a single mesh that has been boolean added together). The pillar I moved out of position is non-manifold.  When the slicer tries to make it manifold (basically capping ends here to make it _solid_) it is collapsing some geometry in the process. 

[image]

mcboeing, 2019-10-05:
JohnOCFII and Jebtrix:
My great thanks to the both of you for taking a look at this problem.
I'll definitely be digging into the comments you have left to see what happened.
Part of the problem is I've been using Fusion 360/Slicer/Prusa printer combination for about two years. And with over 100 designs/prints I've had great success.  So when one pops up with errors was in the dark.
Thanks, again.  So nice to see people willing to help.
Dave

Jebtrix, 2019-10-05:
> Part of the problem is I've been using Fusion 360/Slicer/Prusa printer combination for about two years. And with over 100 designs/prints I've had great success. 

I've seen software costing 25K+ produce questionable STL exports. It happens. I still get some ugly meshes from SolidWorks here and there depending on how a part is built. Fusion historically had some STL export issues of the mesh with fillet feaures. Checking to see if a model can be split into parts/objects in PrusaSlicer is an easy clue to problems may be lurking ahead.

mcboeing, 2019-10-06:
$25K?  Ouch, now I don't feel so bad.  But still incredible that, with
little training, I can use Fusion 360 - free - Slicer - free -  a
relatively low-cost printer to achieve great results.  Unfortunately, as in
yesterday, when things go wrong, I'm left with few options.  Thanks, again,
Jebtrix!

If anything, I'll start taking a deeper dive into 3D printing tech - and
splitting models, too.

Enjoy the rest of the weekend.

Dave

Kachidoki2807, 2019-10-06:
Hi @mcboeing,
In Fusion 360, before to export as STL, make sure to export solid bodies, and not surfaces. Moreover, you'd better to use the "combine" feature to make only one solid body as a last step. That will be easier for the slicer to work with.
See ya.

mcboeing, 2019-10-06:
Hi, Kachidoki:

Hey, thanks for the tip.  There is SO much information on Fusion 360 it's
easy to miss some of the basics.  Will use the combine in the future.

Thx

closed 2026-09-29 by prusaslicer-bot[bot]