prusa3d/PrusaSlicer #3037: Unwanted bridging layers

domtisdell, 2019-10-05:
### Version
Version 2.1.0+win64
PrusaSlicer-2.1.0+win64-201909160915.zip

### Operating system type + version
Win 10 64-bit

### 3D printer brand / version + firmware version (if known)
Creality CR10S S5 + TH3D_UFW_U1.R2.16a

### Behavior
- Unexpected bridge infill is created (see attached snapshots for Layer 1 and Layer 2)
- Import STL model, slice with 0% infill, 15 x bottom horizontal shell solid layers
- Expected Results: no bridge infill - expected 15 x solid infills
- Actual Results: 1 x solid infill layer followed by 14 x bridge infill layers

#### Project File (.3MF) where problem occurs
[image]

[image]

[Wing Extension.zip](https://github.com/prusa3d/PrusaSlicer/files/3693782/Wing.Extension.zip)

Jebtrix, 2019-10-06:
I only found two ways to fix this and I can't quite connect the dots of how it results in bridge infill with your settings.

* Change extruder Nozzle Diameter from 0.3 to 0.33
* Change Bridge Flow Ratio from 0.4 to 0.45

domtisdell, 2019-10-06:
Fantastic. Thanks for the tips, seems to have worked.  Appreciated.

TD5023, 2020-05-10:
I'll add to this and say that the workaround mostly does the trick, but for some reason, it still slices the top layer of solid infill as bridge, as shown. It's way better than having every layer be bridge, but I still hope that this can be remedied in future versions. I'd like to be able to use a very small number (~.25 or even less) for some flexible materials.
[image]
[image]

closed 2026-09-29 by prusaslicer-bot[bot]