prusa3d/PrusaSlicer #3038: Properties unreadable, Windows 10,  dark themes.

promacjoe, 2019-10-06:
### Version
_Version of PrusaSlicer used goes here_ 
2.1.0 Win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10, fully updated.

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Program UI problem, Not related to 3D printer.

### Behavior
* _Describe the problem_  Properties with unreadable, when using Windows 10 dark themes.
* _Steps needed to reproduce the problem_ Windows accessibility options, dark themes.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_ System should be readable for all individuals, including the visually impaired.
If you are going to use system font colors, use system background colors as well. It's a stupid mistake that Too many programmers make.

* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-10-06:
I see no affect of Dark mode in Windows 10 with PrusaSlicer. Can you provide a screenshot and possibly what settings you changed in Windows?

promacjoe, 2019-10-07:
Let me guess, that's what I would have to be doing if I wanted to use your program. Since no explanation of any of these settings are readable on my computer.
I'm not wishing to sound sarcastic, but it is very annoying to try to use a program, That the text is the same color as the background. I am not a programmer, but I have done some programming. And I know how easy this mistake can be made. But if you're legally blind like I am, you would know how often this situation comes up. And it is frustrating. If you are going to use system colors for your fonts, use system colors for your backgrounds.

[image]
[image]
[image]
[image]
[image]
![Image1](https://user-images.githubusercontent.com/55812301/66283487-4fe3bf00-
e889-11e9-9bcd-ca8a3bb92f62.png)
[image]
[image]
[image]
[image]

Jebtrix, 2019-10-07:
That is accessibility High Contrast Black. I wouldn't consider that a Dark theme in the traditional sense. You are the first to bring it to the attention of the devs. They have already dealt with colorblind problems in the past #1986 so they are receptive to importance of accessibility. I see nothing in my question that provoked such a tone. Devs here are pretty great, I'm sure they will handle this with the consideration it deserves.

promacjoe, 2019-10-07:
I'm sorry if I sounded sarcastic, but my frustration for this type of problem is showing. I am using one of the standard Windows 10 accessibility themes. There is nothing special about it. But this has been a ongoing problem for years. This problem even exist in some programs designed for the Partially blind. And my complaints seem to be ignored. Sometimes I think it will take an act of Congress to fix this problem, but they seem to be too busy conducting a witchhunt. Again my apologies. And I appreciate any help you can give.

F3fxcG53dzv, 2020-03-20:
Any news from devs? There is more people (including me) who needs to use windows high contrast theme. I am partially visual impairment user and white&light backgrounds are kryptonite for my eyes. Currently Prusaslicer is nightmare to use and it is big disappoint when you have Prusa printer but cant use prusaslicer. 

I have seen some pics from Prusaslicer mac version which supports dark mode and it looks good. I wish we get that to Windows version soon. Or better support for windows high contrast themes.

bubnikv, 2020-03-20:
Sorry, but there is no support from Microsoft for Win32 and dark mode. Once
Microsoft implements Win32 dark mode support, we will implement support
into PrusaSlicer.

I am sorry, but implementing the high contrast mode is much more work than
it seems, we will not do that for PrusaSlicer 2.2.0 release.

F3fxcG53dzv, 2020-03-20:
Ok, sorry to hear but i keep fingers crossed. 

Btw, older SlicerPE versions works well with high contrast themes, what is changed?

bubnikv, 2020-03-20:
I am not sure, but we had a hard time implementing support for high resolution displays. There was a huge amount of changes around the UI code since Slic3r PE.

F3fxcG53dzv, 2020-03-20:
Ok, thanks for your replies. It might be better to wait Microsoft dark mode APIs and then make dark mode available which overrides high contrast theme. Dark mode is not that good as high contrast theme is but dark mode is thousand times better than white backgrounds. 

In best case there should be both, dark mode and support for high contrast themes and visual impairment (and other users too) user can select from app settings which one want to use. But i am happy if either option is supported in near future.

fernandex00, 2020-06-07:
Hi, is there any update for this?I can't find any info about win32 apps theme support from windows

Kollimolli, 2020-10-10:
For me it looks like a Prusa devs are bit lazy when blaming Microsoft about that API. It is not that hard to create an option for a dark theme or studying and try what windows high contrast theme is and fine tune colors to for it. Even a temporary dark theme would be nice until Microsoft fixes their dark mode APIs, if that never happens.

slohmaier, 2021-03-04:
I use the function: wxSystemSettings::GetAppearance().IsDark()
https://github.com/slohmaier/PrusaSlicer/commit/0930d93f1c825caffa508ae4ef806f48a19554b7
This works for HighContrast white and High Contras Black Windows themes:
[image]
[image]
I'm a newbie to wxWidgets and PrusaSlicer. Is this flag good enough?

klimlik, 2021-04-05:
Whoa! Thank you for the solution it will really helps me a lot!
---
Generally speaking:
I wonder why in the world Prusa developers couldn’t do it if fixing the problem was that “simple”? But forgive that because we finally have solution now :)

YuSanka, 2021-04-07:
Thanks for your reports and suggestion(@slohmaier). High Contrast themes are fixed now in PrusaSlicer with https://github.com/prusa3d/PrusaSlicer/commit/effad844e2f38bddba2a81524dd8cae6572207d4 and will be a part of next release

Vlkvo, 2021-07-09:
> Thanks for your reports and suggestion(@slohmaier). High Contrast themes are fixed now in PrusaSlicer with [effad84](https://github.com/prusa3d/PrusaSlicer/commit/effad844e2f38bddba2a81524dd8cae6572207d4) and will be a part of next release

Is this fix added soon for Prusaslicer?

MarkNahabedian, 2021-08-09:
I am experiencing this problem with PrusaSlicer Version 2.3.3+win64.

Labels are invisible under the "High Contrast Black" theme.  Also, I can't see how to exit the settings screen so I quit PrusaSlicer entirely.

What do I need to do to get visible settings dialogs?

bubnikv, 2021-08-11:
@MarkNahabedian Please post a screenshot.

bubnikv, 2021-08-11:
@YuSanka believes the high contrast theme has been fixed in master with https://github.com/prusa3d/PrusaSlicer/commit/effad844e2f38bddba2a81524dd8cae6572207d4
The fix will be released with PrusaSlicer 2.4.0-alpha1.

Also we now support Dark Mode on Windows, see https://github.com/prusa3d/PrusaSlicer/issues/4493

closed 2021-08-11 by bubnikv

MarkNahabedian, 2021-08-12:
Thanks.

Until the fix has been released, how do I move forward?

I installed a profile for the Monoprice Mini.  That profile appears in the printers menu when I run PrusaSlicer, but I can't find any file under my PrusaSlicer installation directory containing "Monoprice".

I assume that if I could find the file where PrusaSlicer stores the profile for the Monoprice Mini, I could edit it directly and restart PrusaSlicer.