prusa3d/PrusaSlicer #3039: Can't turn off z hop

Sidic101, 2019-10-06:
### Version
2.1.0

### Operating system type + version
Windows 10, latest update

### 3D printer brand / version + firmware version (if known) 
Irrelevant

### Behavior
The lift z option in retraction settings reverts to previous value automatically when I set it to 0

Jebtrix, 2019-10-06:
Try pressing **Enter** after changing to 0

Jebtrix, 2019-10-06:
Duplicate of #3031

Sidic101, 2019-10-06:
It also reverts when I press enter.

Jebtrix, 2019-10-06:
I had to play around with editing the textbox value repeatedly but I did eventually get it to refuse to take 0 once. Hard to find a rhyme or reason. Have you tried 0.0 or making the current value ie 1, to 1.0 then to 0? I know it seems odd but also try 0.000001 then 0.

Sidic101, 2019-10-07:
If I change it to 0.1 and than 0, it works.

lukasmatena, 2019-11-08:
This is a duplicate of #3108. It should be solved now, the fix will be in PrusaSlicer 2.2.0.

Closing.

closed 2019-11-08 by lukasmatena