prusa3d/PrusaSlicer #3040: [Feature Request] Replacing models with new STL

acorderob, 2019-10-06:
### Version
2.1.0

I would like the option to replace models with new versions without disturbing other attached parts like support enforcers/blockers or other model options. Just replacing the STL.

This would simplify a lot printing revisions of a model while iterating.

bubnikv, 2019-10-08:
I believe your request is a duplicate of #552 #1471 #1890

acorderob, 2019-10-08:
No, it's not the same. Those other issues relate to the "Reload from disk" option, which seems to just reload the model from the saved 3mf file.

What i want is to, either reload from the original stl file (which would be problematic because the original path would have to be saved to the 3mf), or just ask for a new stl.

acid-zero, 2019-10-12:
A model replace feature would be great. When designing and saving in Fusion360, Fusion automatically appends "v#" into the end of the filename where # represent the save version. There are times you want to insert that new version model into an existing (complex) 3mf file (replacing the older version of the stl with the newer version) in PrusaSlicer without having to rebuild manually all the modifiers. You don't want to have to rename the stl (to the old version's name) and use a "reload from disk" option, as then you lose track of the stl's build version.

grosserj, 2019-11-21:
A simple workaround is to:

1. right-click on the part
2. Add Part -> Load...
3. Select the updated STL
4. Position the new object in the local coordinates to X:0 Y:0 Z:0
5. Delete the old object

This method keeps the modifiers and layer heights.

bubnikv, 2020-03-15:
please test the latest reload from disk feature in PrusaSlicer 2.2.0-rc3.

closed 2020-06-21 by acorderob

mythmon, 2021-08-23:
Using PrusaSlicer 2.3, I don't see a way to replace a model with a _different_ STL file. Reload from disk is not what this issue was about, as acorderob explained. Can we re-open this, or point to another issue that covers replacing a model on the print bed with a _different_ STL file?