prusa3d/PrusaSlicer #3041: Crash when arranging group of object instances

vintagepc, 2019-10-06:
### Version
2.1.0

### Operating system type + version
Linux (OpenSuSE Leap 42.3, x64)

### 3D printer brand / version + firmware version (if known)
Mk3S/MMU2, FW 3.8.0

### Behavior
Selecting a group of objects and pressing Shift+A to arrange them causes a crash. This is particularly annoying when arranging in MMU mode as one often does not want to arrange the wipe tower as part of the object set.

Steps to reproduce: Open attached 3MF
Select the objects group by selecting "instances" in the right hand list, or the parent object, to exclude the wipe tower.

Press shift+A

Result: Segfault. 
Expected: Arrangement of selected objects.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[tens_slide_bearing_-_tens_slide_bearing.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/3694368/tens_slide_bearing_-_tens_slide_bearing.3mf.gz)

Jebtrix, 2019-10-06:
This does not crash with latest git build so looks to be fixed.

bubnikv, 2020-03-15:
thanks, hopefully fixed, closing.

closed 2020-03-15 by bubnikv