prusa3d/PrusaSlicer #3042: Arranging objects does not always account for support structures

vintagepc, 2019-10-06:
### Version
2.1.0
### Operating system type + version
OpenSuSE 42.3 (x64, Linux)

### 3D printer brand / version + firmware version (if known)
Mk3S/MMU2

### Behavior
Arranging objects when in soluble support mode does not account for first layer support structures. This causes toolpaths to overlap and potentially have print issues.

Expected: Adequate spacing between objects. 

Screenshot: (The lone object on the right was separated to show the actual support footprint)

[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs:
See zip attached to #3041, same 3MF applies.

Jebtrix, 2019-10-06:
Related to #1275

Russkil, 2024-01-05:
+1 for this change to autocalculate the distance

vintagepc, 2026-09-02:
Confirming the default spacing for arrange in 3.0 still seems too low for traditional grid support.  Boosting it to 7 or 8 when support generation is enabled seems to handle the referenced test case well for all 3 support styles. Though I can understand that "set it manually" it might be considered an acceptable workaround now that the spacing feature/setting is more accessible since 2.1 when this was filed and the conflicting toolpath warning does show for this case. 

That said, it seems like it should be possible to detect if the conflicting paths are automatically generated support (the sliced preview knows they are to colour them green) and if so explicitly offer to rearrange with a higher spacing value.