prusa3d/PrusaSlicer #3045: Temp management error with filament switch

mbordoni, 2019-10-07:
### Version 2.1.0+win64
### OS Windows 10 Professional 64bit
### 3D printer Prusa MK3S MMU2S 3.8.0

### Behavior
* When different filament with different profiles are used. During Filament switch the temp for the new filament is changed before unloading the current filament.
* Generate GCode using different filaments with different nozzle temperatures.
Print it and verify nozzle temp switch during cooldown movements on wipe tower. 
* _Expected Results_ the temp has to change after the unload of the current filament and before loading the new one.
* _Actual Results_ the temp is changed before the unload of the current filament so the filament is uloaded with the wrong temperature.
  

_Is this a new feature request?_ No it is a bugfix for MMU2S support

JohnnyDeer, 2019-10-07:
Duplicate of #2971 #2418

closed 2020-08-31 by rtyr

mbordoni, 2020-08-31:
Did it by mistake, sorry.