prusa3d/PrusaSlicer #3047: How can I download the system presets for PrusaSlicer2.1, there don't seem to be any?

rogerfroud, 2019-10-07:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-10-07:
Those are not default nozzle sizes provided as defaults. Your going to need to make them which is relatively simple using tweaked copies of closest Prusa default profiles. When you use ie 0.25 Print Setting profiles to create comparable 0.20 profile versions make sure you pay attention to Dependencies > Compatibility printers condition setting.
`printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK3.*/ and nozzle_diameter[0]==0.25`
would need to be changed to 
`printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK3.*/ and nozzle_diameter[0]==0.2`

If you miss that it would only show up when 0.25mm nozzle Printer Profile was selected instead of 0.20.

Specific Prusa3D settings (or lack of) questions may be best answered here:
https://github.com/prusa3d/PrusaSlicer-settings

Jebtrix, 2019-10-07:
When your in the configuration wizard did you select all available nozzle sizes for your model?

Jebtrix, 2019-10-07:
I can't speak on why they don't include more nozzle sizes. That would be a question on PrusaSlicer-settings. I can guess the answer but that is no help here.

rogerfroud, 2019-10-07:
I've gone back into the wizard and now I can see the 0.25mm and 0.6mm ones.
That's not much help when they're not the sizes I have though.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Jebtrix, 2019-10-07:
I'll give you a jump start.

Extract, then File > Import > Import Config Bundle
[PrusaSlicer_config_bundle-0.2-0.3_MK3S_MMU2_Single.zip](https://github.com/prusa3d/PrusaSlicer/files/3699945/PrusaSlicer_config_bundle-0.2-0.3_MK3S_MMU2_Single.zip)

Jebtrix, 2019-10-07:
Here is a jump start. You may want to add fast type print profiles which mainly increases infill speeds. 

[PrusaSlicer_config_bundle-0.2-0.3_MK3S_MMU2_Single-rev2.zip](https://github.com/prusa3d/PrusaSlicer/files/3699967/PrusaSlicer_config_bundle-0.2-0.3_MK3S_MMU2_Single-rev2.zip)

rogerfroud, 2019-10-08:
Many thanks, that's a big help.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

bubnikv, 2020-02-04:
Could we close this issue then?

closed 2020-02-04 by bubnikv