prusa3d/PrusaSlicer #3048: Configuration wizard doesn't include MK3SMMU2S preset for .25 nozzle

gbelfor, 2019-10-07:
### Version
Version: 2.1.0+win64
Build: PrusaSlicer-2.1.0+win64-201909160915

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 12,759MB

### 3D printer brand / version + firmware version (if known)
10/3/2019 6:11:53 PM: 
slic3r version: 2.1.0+win64
print: 0.15mm QUALITY MK3
filaments: Prusament PLA MMU2
printer: Original Prusa i3 MK3S MMU2S
vendor: PrusaResearch, version: 1.0.3, min slic3r version: 2.1.0-alpha0
model: MK3SMMU2S, variants: 0.25, 0.4, 0.6

### Behavior
Go into configuration wizard.
Select the checkboxes for all three nozzle sizes under Original Prusa I3 MK3S MMU2S
(don't check any other selections)
Complete with NEXT, NEXT, NEXT, FINISH
Then look at the dropdown for "Printer" on the Plater page.

... I see only 2 selections for Original Prusa I3 MK3S MMU2S (0.6 and the standard .4)
... I see all three selections for  Original Prusa I3 MK3S MMU2S Single

[image]

Not a new feature

#### Project File (.3MF) where problem occurs
No .3MF needed

Jebtrix, 2019-10-07:
I answer how its possible to do this here:
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/configuration-wizard-wont-add-25-nozzle-configuration-for-mk3smmu2s/

Jebtrix, 2019-10-07:
As for why this is https://github.com/prusa3d/PrusaSlicer-settings would be the place to inquire.

rtyr, 2019-10-08:
0.25mm nozzle variant is available only for single mode. It is intentional. It is possible that we will add 0.25mm MM variant in the future, but current volumetric speeds used in ramming sequence might be too high for 0.25mm nozzle.

Jebtrix, 2019-10-08:
@rtyr Ah, I figured as much. Thanks for the info.

gbelfor, 2019-10-08:
Thanks, @rtyr.   One question: After reading the Prusa Blog post here: https://blog.prusaprinters.org/everything-about-nozzles-with-a-different-diameter/, I figured it was supported on the MMU. Now I wonder if that business card printed with the 0.25 nozzle was done by manually swapping filament for each color layer.  The blog post didn't explain this...is this how it was done?

rtyr, 2019-10-08:
It was actually printed on regular MK3 (without MMU unit) using standalone Colorprint application (manual filament change at selected layer). This feature is now integrated into PrusaSlicer. You can insert “pause for filament change” gcode command at layers which you marked with the layer selection slider in the print preview screen. This feature can be used only with "MK3S MMU2S Single mode" printer profile at the moment (it is still manual filament change).

Check this video: https://youtu.be/bzf20FxsN2Q?t=361 or visit https://blog.prusaprinters.org/color-print/

  
[image]

closed 2020-04-22 by rtyr