prusa3d/PrusaSlicer #3049: Feature request: Temporarily save configuration state on program exit.

domesticatedviking, 2019-10-08:
(ALL VERSIONS)

Almost every time I close PrusaSlicer, I encounter the following dialog box.

[image]

This box doesn't communicate what has been changed very well, and resolving it is an irritating multi-click process that I usually don't want to deal with when I'm closing the program.

My preference would be for the current configuration state to be temporarily written to disk on program exit, then reloaded automatically when the program restarts.    The interface could signal profiles that had unsaved changes via tab highlighting or similar.   
This would have the added benefit of preventing the need to re-create unsaved profiles after a program crash.

This behaviour could be turned on and off from the preferences menu.  It could also be triggered by an additional "SAVE CURRENT STATE" button on the exit dialog box.

Thanks for considering this.

kumpuu, 2019-10-22:
Im working on a dialog that shows exactly what has changed right now. And I will consider your autosave suggestion. Stay tuned 😀

domesticatedviking, 2019-10-22:
Thanks!  It would be great if that dialog had buttons for saving the states if the user wanted to save them.   Providing the information is good, but having more detail means more things to remember if I still have to navigate to multiple tabs to store these changes.

kumpuu, 2019-10-22:
Yup thats exactly what I had in mind. And letting you quick edit everything but that may be a bit too much work to figure out how to do for me.

Iqwertz, 2020-08-14:
When is this feature available? The current behavior is really annoying and keeps me from fully switching to prusaslicer!

bubnikv, 2020-08-15:
@YuSanka is working on such feature right now.

bubnikv, 2020-10-16:
Implemented in PrusaSlicer 2.3.0-alpha1, based on the UI desing and UI code by @kumpuu , thanks.
Closing.

closed 2020-10-16 by bubnikv