prusa3d/PrusaSlicer #3050: Global settings for Print Host

kml-pt, 2019-10-09:
### Version
2.1.0-win64-201909160915

### Operating system type + version
windows 10 build 18362

### 3D printer brand / version + firmware version (if known)
MK3S/MM2S
3.8.0/1.0.6

### Behavior
I would really like to be able to set my Print Host settings one time, and have them apply to all printer settings. It's not possible to save into the system presets, so far as I know, and it is very inconvenient to have to update print host settings for a bunch of different printer setups every time the system presets are updated.

_Is this a new feature request?_ Yes

Jebtrix, 2019-10-09:
Currently the workaround to do that is changing the the vendor\PrusaResearch.ini

Find these keys and change to your desired values
```
octoprint_apikey = 
octoprint_host =  
```
If a online profile update is done you'll have to redo this.

kml-pt, 2019-10-10:
Thank you, I will try that.

bubnikv, 2020-10-17:
PrusaSlicer 2.3.0-alph1 separates the Physical printer settings from the old logical printer profile.
Closing.

closed 2020-10-17 by bubnikv