prusa3d/PrusaSlicer #3051: Bundle a custom printer settings on installer

italocjs, 2019-10-09:
Hello, sorry for creating an issue for this, but i didnt find much information about it,  I am a printer manufacturer a i really want to ship prusa slicer instead of cura on my printers,   I managed to make a installer that "copy paste" the %appdata% folder with the settings, but i'm quite sure there is a better way to embed this profile (and also to update the software and profiles as prusament materials does).  

I'm having a problem using the %appdata% copy, because on multiple users, only one (the one that installed the program) get the settings.

Anyone has any idea? Thanks in advance

neophyl, 2019-10-09:
Why don't you just create a Config bundle like most people do to back up all their settings/profiles ?  Its under the file menu.  Once you have a Config bundle saved it can be imported into Prusa Slicer at any point later.

italocjs, 2019-10-09:
I tried, but it fails when importing (says corrupted file), probably a bug in the version i'm using (latest stable 2.1.0).  Tried reinstalling a few times and on another computer.

neophyl, 2019-10-09:
Then it sounds like you have other issues to worry about.  If saving/importing a config bundle doesnt work have you tried just saving and importing a single profile as a test ?  

Would you send me your failing config bundle ?  I can at least try it and have a look.  Might not be able to figure out why it wont import but it cant hurt to try :)  Just zip it up and attach it here.

Jebtrix, 2019-10-09:
As a manufacturer for maximum flexibility I would use \vendor\MyCustomPrinters.ini. Because it uses inheritance you would be able to change values there and any new profiles users make would inherit the changes automatically. It would create true non changeable defaults that the user would have to make copies of to customize. When you update your vendor ini, those would all update. Any new profiles the user makes from them get the new values. The UI would also show (with the lock icon) with existing older profiles that values are different than factory defaults.

Jebtrix, 2019-10-09:
Almost forgot, by using vendor ini you can also set your own config_update_url to allow automatic checking and updating of profile online.

Jebtrix, 2019-10-09:
@italocjs You may want to start a settings repo for the printer(s). It would be easier for community to help out. You can use https://github.com/prusa3d/PrusaSlicer-settings as reference.

italocjs, 2019-10-10:
Hello, sorry for the delay.

@Jebtrix  this is exactly what ive been looking for, thank you so much!!!

@neophyl , figured it out, it was a latin character describing the gcode (example ã and õ), i removed it worked perfectly, i'm attaching the ini for future reference if needed.  Thanks for the help!

[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/3712157/PrusaSlicer_config_bundle.zip)

closed 2019-10-10 by italocjs