prusa3d/PrusaSlicer #3054: [FEATURE REQUEST] - Adjust Speed at layer height Editing Option

ondrovic, 2019-10-10:
### Version
2.10+win64

### Operating system type + version
Windows 64-bit Pro (build 18362)

### 3D printer brand / version + firmware version (if known)
MK3S
FW: 3.80

### Behavior
* Load Model
* Click on the Editing Option
  * Select Option

New Feature Request: Yes

#### Project File (.3MF) where problem occurs
No problem to report

[FeatRequest_SpeedAtLayers.zip](https://github.com/prusa3d/PrusaSlicer/files/3713729/FeatRequest_SpeedAtLayers.zip)

Jebtrix, 2019-10-10:
It's already there?
[image]

Jebtrix, 2019-10-10:
Note it requires being in Expert Mode to see those choices.

ondrovic, 2019-10-11:
Damn yep missed it thanks

closed 2019-10-11 by ondrovic