prusa3d/PrusaSlicer #3055: Consistent crashes when slicing attached object

mdaneman, 2019-10-11:
### Version
2.1.0

### Operating system type + version
Windows 10 version 1903
OpenGL installation
GL version: 4.6.0 NVIDIA 416.16
Vendor: NVIDIA Corporation
Renderer: GeForce GTX 1070/PCIe/SSE

### 3D printer brand / version + firmware version (if known)
MK3S, Firmware 3.8.0

### Behavior
When slicing using specific set of settings (Flex filament) and possibly specific model, Slicer crashes and exists. Note that slicing the same object with the opposite face facing down, it does not crash with the same settings.

Load attached .3mf file and hit "Slice" to reproduce.

* _Expected Results: expected it slice and not crash.

* _Actual Results: Slicer crashed and exited during slicing.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request? No

#### Project File (.3MF) where problem occurs
[Prusa Leg Adaptor v6.zip](https://github.com/prusa3d/PrusaSlicer/files/3716025/Prusa.Leg.Adaptor.v6.zip)

Jebtrix, 2019-10-11:
Duplicate of #2988

Jebtrix, 2019-10-11:
A temporary workaround is to set Filament Settings > Filament Overrides = 0.0001

bubnikv, 2020-03-15:
fixed  on Nov 22, 2019 with 6fcd51d
Closing

closed 2020-03-15 by bubnikv