prusa3d/PrusaSlicer #3056: Wrong last layers

espr14, 2019-10-11:
v2.1.0 win64

Should look like this:
[image]

But it's sliced with extra 2 layers (1 perimeter and bridges):
[image]

Source:
[wrong top.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3716116/wrong.top.3mf.zip)

smartroad, 2019-10-11:
I opened up the 3mf file and it has auto repaired 104 errors in the mesh. More than likely there is a hole in the mesh or a stray line inside it somewhere. What software did you use to design it?

espr14, 2019-10-11:
OpenSCAD.

espr14, 2019-10-11:
The STL generated by OpenSCAD is bad, the one generated by FreeCAD is OK. Still the preview does not correspond to the sliced result.

espr14, 2019-10-11:
I have fixed the model by making parts to overlap to create single body instead of many bodies touching each other. Now both preview and sliced result are OK. Still, the original model should be repaired well or the sliced result should correspond to the preview or it should notice user that there are unrepairable errors.

Jebtrix, 2019-10-11:
I wouldn't consider this a bad mesh. It's manifold. There is some zero faces and edges but just merging duplicate vertices slices fine. @bubnikv Another merged vertices to slice correctly.

[wrong.top_merged_vertices.zip](https://github.com/prusa3d/PrusaSlicer/files/3718889/wrong.top_merged_vertices.zip)

FidelCapo, 2021-02-01:
I can't test the fix for this model because there is no stl file available, only a 3mf file. However, according to tests of other models, this issue should be corrected in next release.
Closing.

espr14, 2021-02-01:
You can export STL from the 3MF.

neophyl, 2021-02-02:
The act of saving the file as a 3mf can actually fix certain errors of the model in some cases. As such exporting from the 3mf is not guaranteed to exactly match the original model. 
I think they want as many of the original models as they can get to properly test the changes.

espr14, 2021-02-02:
After so long time I doubt I can find the original STL. However, PS 2.3.0 repairs it well so no reason to keep it opened.

closed 2021-02-02 by espr14